mavlink/qgroundcontrol #10071: Gstreamer: internal data stream error

new-user-name, 2021-12-11:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Test Gstreamer pattern should be visible

## Current Behavior
Test Gstreamer pattern is not visible on Android and visible on Windows

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior

1. Build QGC from master, no errors, Windows 10 and Android (armeabi-v7a).
2. Gstreamer test
 gst-launch-1.0 videotestsrc ! x264enc ! rtph264pay ! udpsink port=5000 host=127.255.255.255
for Windows build, OK
3. Gstreamer test
 gst-launch-1.0 videotestsrc ! x264enc ! rtph264pay ! udpsink port=5000 host=192.168.0.103
for Android build, GstVideoReceiver.cc:1305 - Gstreamer error: internal data stream error".

I tried Android builds on Windows and Ubuntu, the result is the same.
Beta form Playmarket gives the same error.

Android 6.0, ARM-V7.

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

new-user-name, 2021-12-11:
Gstreamer warning level gives a lot of
`<h264parse0> broken/invalid nal Type`
both in Windows and Android,

kressinluiz, 2021-12-12:
You can try to:
- enable low latency mode
- change the caps after x264enc to ! video/x-h264, stream-format=avc ! 
- set is-live property of videotestsrc to true

new-user-name, 2021-12-12:
Low latency enabled already, no avail. I can't change Gstreamer test because my goal is to make Android work with exactly the same Gstreamer string as in Windows.

new-user-name, 2021-12-12:
So I can change onlу receiver Gstreamer config.

new-user-name, 2021-12-12:
God knows why, the thing works on Android 7. Maybe I can find additional logs somewhere?..

kressinluiz, 2021-12-12:
Currently I see the qgroundcontrol pipeline as (udpsrc):
[image]
You can try this pipeline that imitates qgroundcontrol pipeline (udpsrc):
- gst-launch-1.0 --gst-debug 1 udpsrc address=localhost port=5600 ! application/x-rtp, media=video, clock-rate=90000, encoding-name=H264 ! parsebin ! tee name=t ! queue ! valve ! decodebin3 ! autovideosink t. ! queue ! valve ! matroskamux ! filesink location=teste05.mkv
(matroskamux can be changed for mp4mux or qtmux)
It is easier to debug from the command line. You can change the gstreamer debug level but it is hell to find any useful info.

If I try:
- gst-launch-1.0 videotestsrc ! x264enc ! rtph264pay ! udpsink port=5000 host=localhost
It doesn't work. It returns internal data stream error.  But I can make it work with:
- gst-launch-1.0 --gst-debug 1 videotestsrc pattern=ball is-live=true ! video/x-raw, format=I420, width=640, height=480, framerate=30/1 !  x264enc ! rtph264pay ! udpsink host=localhost port=5600

So I think if you look into the errors that popup when you remove is-live=true and video/x-raw caps, you can find what is missing from the receiver pipeline.

new-user-name, 2021-12-13:
The second example works, you are right. Just change port=5000 to 5600 in first short example (the main point stays the same).

new-user-name, 2021-12-13:
Test could be reduced to
gst-launch-1.0 --gst-debug 1 videotestsrc pattern=ball ! video/x-raw, format=I420 ! x264enc ! rtph264pay ! udpsink host=localhost port=5600

closed 2021-12-13 by new-user-name