mavlink/qgroundcontrol #10087: Waiting for video, cannot display IPcamera RTSP video stream

TyphoonWu, 2021-12-18:
## Expected Behavior
Display RTSP video stream.

## Current Behavior
Waiting for video, cannot display RTSP video stream.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. IPcamera enable RTSP, RTSP mount points: rtsp://10.0.2.145:554/11 and rtsp://10.0.2.145:554/22. The address include video and audio.
2. Qgroundcontrol video steam set RTSP: rtsp://10.0.2.145:554/11
3. Return to fly/main UI

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: v4.1.1(latest release version)
- Flight Controller: Pixhawk 2
- Autopilot (with version): Ardupilot Copter 3.6.1

## Detailed Description
1. I have one device IPcamera, IP camera enable RTSP, RTSP URL: rtsp://10.0.2.145:554/11 (for 720P) and rtsp://10.0.2.145:554/22 (for 640P). The address include video and audio. Qgroundcontrol RTSP video steam set the two RTSP url both cannot display the video stream. 
2. Gstream display video success, command: gst-launch-1.0 rtspsrc location=rtsp://10.0.2.145/22 latency=0 ! queue ! decodebin ! autovideosink
3. Misson planner display video success, command: rtspsrc location=rtsp://180.188.18.71:42559/11 latency=0 ! queue ! application/x-rtp ! rtph264depay ! avdec_h264 ! videoconvert ! video/x-raw,format=BGRA ! appsink name=outsink
2. Use raspberry camera's RTSP  it's ok.

## Log Files and Screenshots
- [QGC Console Logs](
[image]
)
- Doesn't related with Autopilot. Failure screen shot:
[image]
[image]

- Use gstream command it's ok to display RTSP video stream:
[image]
[image]

booo, 2022-01-04:
> I have one device IPcamera, IP camera enable RTSP, RTSP URL: rtsp://10.0.2.145:554/11 (for 720P) and rtsp://10.0.2.145:554/22 (for 640P). The address include video and audio. Qgroundcontrol RTSP video steam set the two RTSP url both cannot display the video stream.

In this example you use a port as far as I can tell. In the other working examples you dont use a port or a different port. Is this intended?

keiler2018, 2022-01-08:
I compiled the master branch, and I encountered the same problem

keiler2018, 2022-01-08:
> I compiled the master branch, and I encountered the same problem

I can't use the command line. Just add a parameter   “--gst-debug-level=3”

keiler2018, 2022-01-08:
rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov 

This is OK, but the camera is not

TyphoonWu, 2022-01-21:
> Sorry, with explain clearly. Actually, It doesn't matter for the port, 
Gstream display video success, below commands both ok: 
gst-launch-1.0 rtspsrc location=rtsp://10.0.2.145/22 latency=0 ! queue ! decodebin ! autovideosink
gst-launch-1.0 rtspsrc location=rtsp://10.0.2.145:554/22 latency=0 ! queue ! decodebin ! autovideosink

The IPcamera have default port 554, the URLs rtsp://10.0.2.145/22 and rtsp://10.0.2.145:554/22 both able to access the video, 
And I use mission plan and VLC to load the URLs both ok.

TyphoonWu, 2022-01-21:
> In this example you use a port as far as I can tell. In the other working examples you dont use a port or a different port. Is this intended?

Sorry, with explain clearly. Actually, It doesn't matter for the port,
Gstream display video success, below commands both ok:
gst-launch-1.0 rtspsrc location=rtsp://10.0.2.145/22 latency=0 ! queue ! decodebin ! autovideosink
gst-launch-1.0 rtspsrc location=rtsp://10.0.2.145:554/22 latency=0 ! queue ! decodebin ! autovideosink

The IPcamera have default port 554, the URLs rtsp://10.0.2.145/22 and rtsp://10.0.2.145:554/22 both able to access the video,
And I use mission plan and VLC to load the URLs both ok.

jingwenyi, 2022-02-15:
if you test in windows, you can change code"gst_element_factory_make("parsebin", "parser") 
 to     gst_element_factory_make("decodebin", "decoder")"   in gstvideoreceiver.cc ;

taileron, 2022-02-17:
with the mac releases (all dailys along 2021/2022) I had very similar experiences there an additional very strange thing happens. If you push the record button, perfect valid video files get recorded while the QGC video window shows "WAITING FOR VIDEO" instead of the content. It can be easily tested with the iOS app periscope hd that streams the smartphones cam directly into rtsp
The streams alway get visible with VLC or QGC 3.xx

https://discuss.px4.io/t/video-stream-to-qgroundcontrol-mac/2327

joecfd, 2022-07-14:
I can see video on android, there is no audio. What is going on.

jingwenyi, 2022-07-15:
please use the version 4.01

joecfd, 2022-07-15:
Thanks for your reply. I checked the pipeline in the source code of version 4.0 and did not see audio elements. That says audio is not available. There is not branch of Version 4.01.

closed 2025-05-03 by HTRamsey