mavlink/qgroundcontrol #10104: Yaw and Throttle not available from joystick in v4.2.0

AeroTH310, 2022-01-06:
## Expected Behavior
All joystick channels should be available for the Logitech Extreme 3D Pro, although not necessarily mapped as desired or calibrated.

## Current Behavior
The yaw axis, throttle axis, and several buttons are not available for mapping or calibration.  The joystick "General" and "Calibration" pages show that these two channels do not change any QGC inputs when moved on the device.  Joystick calibration is impossible.  Also, several of the button channels cannot be mapped.

## Steps to Reproduce
1. Plug in Logitech Extreme 3D Pro
2. Run QGC v4.2.0
3. Connect to a vehicle (I ran SITL using "make px4_sitl gazebo_standard_vtol")
4. Navigate to joystick tab
5. Manipulate yaw and throttle axes while observing channel monitors on the joystick "General" page (also push all the buttons)

## System Information
- Operating System: Ubuntu 20.04
- QGC Version: v4.2.0
- QGC build: cf9588a (AppImage from github)
- Flight Controller: px4_sitl gazebo_standard_vtol
- Autopilot (with version): PX4 commit# 170849c8 (built locally)

## Detailed Description
The issue is not reproducible on v4.0.10 (latest confirmed).  Note v4.1.4 will show movement on the channel monitors on the joystick Calibration page, but not the joystick General page (even after successful calibration).  The issue cannot be rectified using jstest-gtk on Ubuntu.  The issue cannot be rectified by altering .config/QGroundControl.org/QGroundControl.ini.

## Log Files and Screenshots
- Here is a screenshot of the behavior on v4.2.0.  For this shot, I was putting in roughly mid-range inputs to the roll, pitch, yaw, and throttle axes.  Notice that only two channel monitors show any input.
[image]
 - Here is the same circumstance reproduced on v4.0.10.  Notice that the channel monitors show input on all four axes.
[image]

ericjastram, 2022-01-17:
had the same issue with my Logitech Extreme 3D Pro.. my temporary workaround was to disable the check to see if it is a game controller... `if (**false &&** SDL_IsGameController(i)) {
                isGameController = true;`

HTRamsey, 2024-05-31:
This is an SDL mapping issue, and probably has been fixed. Please reopen if not.

closed 2024-05-31 by HTRamsey