mavlink/qgroundcontrol #10107: Problem linking Airmap

Jcwscience, 2022-01-08:
## Expected Behavior
Project build succeeds.

## Current Behavior
Project build fails.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior

mkdir build
cd build && qmake ..
make

## System Information
Sorry, it’s a weird machine.
Nvidia AGX Xavier
Ubuntu impish with weird drivers from Ubuntu bionic.
Aarch64

## Detailed Description
There seems to be an error in the airmap plugin, is there a way to disable airmap support?

```
linking staging/QGroundControl
/usr/bin/ld: skipping incompatible /home/john/HackCopter/qgroundcontrol/build/libs/airmap-platform-sdk/linux/Qt.5.15/libairmap-cpp.so when searching for -lairmap-cpp
/usr/bin/ld: cannot find -lairmap-cpp
/usr/bin/ld: skipping incompatible /home/john/HackCopter/qgroundcontrol/build/libs/airmap-platform-sdk/linux/Qt.5.15/libairmap-cpp.so when searching for -lairmap-cpp
collect2: error: ld returned 1 exit status
make: *** [Makefile:3006: staging/QGroundControl] Error 1
```

booo, 2022-01-10:
Maybe `qmake DEFINES=DISABLE_AIRMAP ..` works for you?

Jcwscience, 2022-01-10:
@booo Perfect that worked! Thanks!

hamishwillee, 2022-04-21:
The best way is to crate  **user_config.pri** in the root of the repo. It should contain the text `DEFINES += DISABLE_AIRMAP`.
Docs being updated now.