mavlink/qgroundcontrol #10121: Joystick gimbal control not moving or really slow.

Tonisvoitka, 2022-01-17:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
When I add a joystick to qgc, gimbal up/down should change the pitch angle. 

## Current Behavior
Gimbal with basecam controller moves slow and only 1 degree

## Steps to Reproduce:
Basic setup with basecam gimbal controller and latest QGC - add joystick, check gimbal down and gimbal up. 

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
The gimbal only moves 1 degree in a really slow manner when using QGC and mavlink commands. 
When I look at mission planner payload control tab, I can see that the gimbal moves from 0.9 to -0.9 degrees. So there must be an decimal place error somewhere.

Tonisvoitka, 2022-02-11:
this is getting frustraiting. This seems to not be a ardupilot issue, because using mission planner payload control sliders the storm32 gimbal controller and the basecam move the gimbal correctly. Only when using QGC and joystick up/down, the values of mount_status pointing_a/b/c differ by 10x. When payload control sliders move it from -4500 to 9000, then joystick with qgc only moves -45 to 90. @DonLakeFlyer please advise, is this still ardupilot issue or something else?

DaKSK, 2022-02-11:
I think the issue is here. The units for MNT_ANGMAX_TIL and MNT_ANGMIN_TIL in parameters are expressed in cdeg, but in Joystick::_pitchStep(int direction) seems to limit the movement in degrees. 

`void Joystick::_pitchStep(int direction)
{
    _localPitch += static_cast<double>(direction);
    //-- Arbitrary range
    if(_localPitch < -90.0) _localPitch = -90.0;
    if(_localPitch >  35.0) _localPitch =  35.0;
    emit gimbalControlValue(_localPitch, _localYaw);
}`

DaKSK, 2022-02-11:
If the pitch movement is limited to -90 to 35 cdeg, then the actual pitch movement would correlate to about a degree.

Tonisvoitka, 2022-02-15:
this issue was solved when 
`void Joystick::_pitchStep(int direction) { _localPitch += static_cast<double>(direction); //-- Arbitrary range if(_localPitch < -9000) _localPitch = -9000; if(_localPitch > 3500) _localPitch = 3500; emit gimbalControlValue(_localPitch, _localYaw); }`

And  Joystick::_executeButtonAction step to 100 in the code

DonLakeFlyer, 2022-02-21:
I'm guessing this is a problem with differing implementations of `MAV_CMD_DO_MOUNT_CONTROL` which is what ends up being sent. PX4 uses degrees and ArduPilot uses centidegrees. QGC currently sends degress in that message. Can anyone verify that?

Tonisvoitka, 2022-02-22:
would it make sense to have it as an option somewhere? Such as when u start QGC it asks allready what system u are using - cant it deteremen Cdeg vs Deg also? It

DonLakeFlyer, 2022-02-22:
It already knows that. Just need to validate what the problem is.

Tonisvoitka, 2022-02-22:
I have used stock QGC with a PX4 autopilot - there the gimbal controller worked, the gimbal moved, but there was another bug, its written in  #9908 
But now, when we changed it to cdeg, it worked with both storm and basecam controllers with latest arducopter versions. 
And there the QGC was changed to PX flightcontroller. 
I do not have a px4 compatible autopilot available to test.

DonLakeFlyer, 2022-02-22:
So is it the autopilot firmware (PX4/ArduPilot) that determines whether degrees or centi-degrees? Or is it the gimbal type/firmware that controls this?

Tonisvoitka, 2022-02-22:
the gimbal controller was the same on both px4 and ardupilot. Gimbal controller only listens to basic mavlink commands. With basecam gimbalcontroller Px4+standard qgc= success, but arducopter+standardQGC=nosuccess. 
Arducopter only worked when we changed deg to cdeg in the code. 
To answer ur question, it seems that the firmware does determine whether to use deg or cdeg. 
No changes were made to the gimbal firmware on both cases.

DonLakeFlyer, 2022-02-22:
Ok, that would be good news thing. @tridge Does this make sense? Autopilot firmware controls these values?

Tonisvoitka, 2022-03-18:
Any updates on this @DonLakeFlyer

DonLakeFlyer, 2022-03-21:
@Tonisvoitka Do you have the ability to build QGC? Can you try the pull I put up?

Tonisvoitka, 2022-03-21:
We will try it. But it has to affect the speed also. Otherwise it will move, but it takes forever

DonLakeFlyer, 2022-03-22:
There is no speed specified in the DO_MOUNT_CONTROL command. No idea where that comes from

Tonisvoitka, 2022-03-22:
We will try it, but last time we did our own build, we had to change Joystick::_executeButtonAction, because it was initially 1 and it moved but really slow. When we changed that to 100 it worked.

DonLakeFlyer, 2022-03-22:
Can you update this pull with that change as well then?

DonLakeFlyer, 2022-03-22:
Or make a new pull based on this one with that as well in it

Tonisvoitka, 2022-04-27:
@DonLakeFlyer We compiled it. and it worked. Solved the issue.

Tonisvoitka, 2022-05-31:
@DonLakeFlyer The only thing that seems to not work is the joystick speed. I have changed it from 10 to 100 and visually it did not affect the speed.

Tonisvoitka, 2022-06-09:
@DonLakeFlyer the speed is not affected. Everything else works perfectly fine. The mnt speed is only visible if we use mission planner and the payload control sliders. There the speed difference can be detected. Otherwise 0 or 100 does not make any difference.

closed 2026-10-04 by github-actions[bot]