mavlink/qgroundcontrol #10124: Android 4.2.0 cannot connect pixhawk5x via usb otg

lfeng17, 2022-01-18:
Android 4.2.0 cannot connect pixhawk5x via usb otg.
Also , using 4.1.4 failed.
Telemetry connection wasn't tested.

vincentpoont2, 2022-01-21:
Tried the following with [QGroundControl Android Version 4.2.0 64bit](https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html#android)

### Pixhawk5X, Durandal, Pix32 V5 :

- Direct connection - **unsuccessful**: 
- Connection via Telemetry - **unsuccessful**

Note: All has USB-C and tried multiple Type-C cables. 

### Pixhawk 4, Pixhawk 4 Mini 

- Direction connection **successful:**  
- Connection via Telemetry - **unsuccessful**

@mrpollo @DonLakeFlyer Any idea?

richtong, 2022-07-28:
Ping, having the same issue with Android 12 to M-robotics Control Zero H7 OEM. It works with Ubuntu fine with a direct connect. We have a USB OTG cable installed but it says disconnectioned

HTRamsey, 2024-05-31:
Android serial support was very limited, I have a PR that updates the drivers and should allow Android to work with a lot more devices.

closed 2024-05-31 by HTRamsey