mavlink/qgroundcontrol #10126: QGC wont start on Windows 11

AlexKlimaj, 2022-01-18:
I am unable to get the release or daily QGC to start on Windows 11. See debug output below.

[image]

AlexKlimaj, 2022-01-19:
Update, 4.0.8 works, 4.0.10 works, anything from 4.1.0 and newer doesn't work.

AlexKlimaj, 2022-01-19:
The only difference in the debug log from a working 4.1.0 is the VideoReceiverLog:Stop called on empty URI

mrpollo, 2022-01-19:
@AlexKlimaj, could this be related to the bump of Qt to v5.15?

AlexKlimaj, 2022-01-20:
I'm not sure. It runs on my other two windows 11 machines. The only difference with this one is that it has dual GPU with the built in Intel and a discrete nvidia.

farhangnaderi, 2022-08-23:
@mrpollo Same here:
[image]

hamishwillee, 2022-08-23:
It works for me on Windows 11 Pro. Today daily build of QGC.

[image]

I have no advice on debugging. Though I would do a very thorough clean of all my data and try again.

Amn04, 2022-09-27:
It is working fine on windows 11. Try deleting your data file of qgc then using it.

AlexKlimaj, 2022-09-27:
> It is working fine on windows 11. Try deleting your data file of qgc then using it.

Do you have dual CPU graphics and a discrete GPU? It wonks on all my single GPU computers but not on the Legion 5i Pro with the intel/nvidia dual GPU.

paoloelle, 2022-10-08:
I solved this problem doing these steps:

1. open `device manager`
2.  disable NVIDIA GPU
3. launch QGC

After this I tried to enable again GPU and QGC seems to work also with the GPU enabled.

AlexKlimaj, 2022-10-08:
Thanks for the tip. I tried disabling the NVIDIA GPU and starting QGC. Then tried disabling the Intel GPU and starting QGC. It didn't work either attempt.

paoloelle, 2022-10-09:
I have the version 4.2.3, I don't know if this further information could be helpful.

kibidev, 2023-03-16:
I checked on a Windows 10 VM, running in "Boxes" on Ubuntu 22.04.
Got similar behavior, 4.2.3 does not work (starts in background, but no GUI), but 4.0.8 works.

MaEtUgR, 2023-03-23:
With QGC itself I never had this issue. I'm constantly updating to daily and it works on my:
- home desktop with i7-8700K GPU unused and GTX 1080 as constant primary.
- work laptop switching between i7-8750H and GTX 1050 Ti with only the scaling being off after switching resolution and window manager scaling

With a drivate of QGC I've had it before on the laptop that some daily versions were only present in the task manager and the window didn't show up but usually switching to a different version, resetting all the AppData and so on fixed it. And I could never see a correlation to the GPU so it might have been something else.

AlexKlimaj, 2023-09-26:
UPDATE: I have a new laptop, a Thinkpad T14s, and QGC won't open again. This laptop only has the built in Intel GPU. Not dual like the Legion.

Same problem on a brand new Samsung Book3 360 laptop

AlexKlimaj, 2023-09-26:
This sounds related.

https://forum.qt.io/post/722023

WilliamDomben, 2023-10-23:
> I am unable to get the release or daily QGC to start on Windows 11 Legion 5i Pro with dual Intel/Nvidia GPUs. See debug output below.
> 
> UPDATE: I have a new laptop, a Thinkpad T14s, and QGC won't open again. This laptop only has the built in Intel GPU. Not dual like the Legion.
> 
> Same problem on a brand new Samsung Book3 360 laptop
> 
> [image]
> 
> [image]

Hello guys, same problem here today 23.10.2023
Qground is running on a brand new laptop with windows 11, got contact with the rov, but we cant seem to get a camera link. Everything else is running fine

Olexandros, 2024-01-25:
I had the same problem, QGC simply wouldn't start. I also checked in the Task Manager, the process was present there. Then I started googling for the answer. While I was doing it the QGC app appeared. Overall, it took ~5 minutes for the first start.

It's weird since I've got a new proper laptop, yet it took ~5 minutes for the first start as if I had an old primitive one. So I suggest to wait for 5-10 minutes and see if the app appears. Most likely it will.

lida2003, 2024-05-09:
> I had the same problem, QGC simply wouldn't start. I also checked in the Task Manager, the process was present there. Then I started googling for the answer. While I was doing it the QGC app [...]

I have QGC latest version, the exact same behavior as yours.

Is it some environment setting related? 

```
Device name	Daniel-SurfacePro
Processor	Intel(R) Core(TM) i5-7300U CPU @ 2.60GHz   2.71 GHz
Installed RAM	8.00 GB
Device ID	C5FC645F-69FA-4372-A8F8-CB8C596953A3
Product ID	00330-80000-00000-AA147
System type	64-bit operating system, x64-based processor
Pen and touch	Pen and touch support with 10 touch points

Edition	Windows 11 Pro
Version	22H2
Installed on	‎23/‎5/‎31
OS build	22621.3527
Experience	Windows Feature Experience Pack 1000.22700.1003.0

```

Olexandros, 2024-05-13:
> I have QGC latest version, the exact same behavior as yours.
>
> Is it some environment setting related?
>
> ```
> Device name	Daniel-SurfacePro
> Processor	Intel(R) Core(TM) i5-7300U CPU @ 2.60GHz   2.71 GHz [...]

I don't think so. The only common thing is that we both use Windows 11.

```
Processor	12th Gen Intel(R) Core(TM) i5-12500H   2.50 GHz
Installed RAM	16.0 GB
System type	64-bit operating system, x64-based processor
```

lida2003, 2024-05-13:
> I don't think so. The only common thing is that we both use Windows 11.

I see.

I had the process in task manager

a) it took for a while (maybe 3~5minutes)
b) and sometime it will never show on desktop, need to kill and restart again

Maybe It's something different.

AlexKlimaj, 2024-08-16:
Fixed by https://github.com/mavlink/qgroundcontrol/pull/11678

closed 2024-08-16 by AlexKlimaj