mavlink/qgroundcontrol #10135: Error: dependent 'C:\qgroundcontrol\custom\src\comm\MockLink.General.MetaData.json.xz' does not exist.

zeev-mindali, 2022-01-23:
# This system is for posting bugs or feature requests ONLY. Support questions will be automatically closed with no response.
# For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
# To post a bug report
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- **If you are using a Stable build which is not the latest, first step is to install latest Stable build and try again prior to entering a bug report.**
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**

using qt 5.15.2 on windows 10 pro with ardupilot.
when compilng the branch v4.2 all is working fine.
when trying compile with custom plugin folder

<code>
Error: dependent 'C:\qgroundcontrol\custom\src\comm\MockLink.General.MetaData.json.xz' does not exist.
</code>

what am i missing??

zeev-mindali, 2022-01-23:
menage to solve it by copying the missing files from src\comm to source\comm

the project finally compiled, but when i trying to launch it, i don't get any screen.
and if i try to relaunch it, it's saying that another instance already opened 

i missing something here, i wanted to create a plugin, but it draw me back every time :( 
please help

zeev-mindali, 2022-01-23:
finally made it work with little help of some python script.

closed 2022-01-23 by zeev-mindali

eyubSalihOzdemir, 2022-10-05:
I am now facing the exact same problems. I also fixed the MockLink.General.MetaData.json.xz error by copying the comm file from main folder to under custom folder.

But I also got no window when I tried to run the custom version.
I get the following errors:
```
QQmlApplicationEngine failed to load component
qrc:/qml/MainRootWindow.qml:468:5: Type FlyView unavailable
qrc:/qml/QGroundControl/FlightDisplay/FlyView.qml:138:5: Type FlyViewMap unavailable
qrc:/qml/QGroundControl/FlightDisplay/FlyViewMap.qml:226:5: Type ObstacleDistanceOverlayMap unavailable
qrc:/qml/QGroundControl/FlightDisplay/ObstacleDistanceOverlayMap.qml: No such file or directory
```

Any idea on how to fix that?