mavlink/qgroundcontrol #10136: Waypoint problem in Chinese and Chinese Simplified display.

YuanFPV, 2022-01-23:
Dear developer,I noticed that in version 4.2.0 32/64 bit of Android qgroundcontrol, when the language is switched to simplified Chinese.Simplified Chinese will also be mixed with traditional Chinese.When I switched the language to Chinese for the first time, everything was normal, but there was a bug when I restarted the software, but English could be used normally.
[image]
[image]
[image]

llaaqa, 2022-02-15:
你好，我的安卓4.2.0无法添加航点

ATLASFROST, 2022-09-07:
你好，我的4.2.0-4.2.3版本都不能添加航点，或者添加航点不显示。Hello, my version 4.2.0-4.2.3 can not add the navigation point, or add the navigation point is not displayed.

closed 2026-10-04 by github-actions[bot]