mavlink/qgroundcontrol #10138: File path wrong can‘t be compiled

Kim-zhi-jiang, 2022-01-25:
Environment: Ubuntu20.04  QT5.15.2  
@Ramón Roche     QGC4.2 and  master try also like this
[image]

In file included from /home/kim/vtol/qgc/qgroundcontrol/src/Airmap/AirMapSharedState.h:17,
                 from /home/kim/vtol/qgc/qgroundcontrol/src/Airmap/AirMapManager.h:12,
                 from /home/kim/vtol/qgc/qgroundcontrol/src/QGCToolbox.cc:38:
/home/kim/vtol/qgc/qgroundcontrol/src/Airmap/services/client.h:4:10: fatal error: airmap/client.h: No such file or directory
    4 | #include <airmap/client.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:190532: QGCToolbox.o] Error 1

Kim-zhi-jiang, 2022-01-25:
@Ramon Roche

Kim-zhi-jiang, 2022-01-25:
@mrpollo

mrpollo, 2022-01-25:
@Kim-zhi-jiang what steps did you follow to get here? please paste all the commands you run, and the full output, try to paste the text please, and avoid a screenshot.

mrpollo, 2022-01-25:
you can also use `DEFINES=DISABLE_AIRMAP make build` and that will skip any airmap configuration

FiddlingRich, 2022-01-25:
> @Kim-zhi-jiang what steps did you follow to get here? please paste all the commands you run, and the full output, try to paste the text please, and avoid a screenshot.

Hi, i have encountered this issue too, i have followed the steps in https://dev.qgroundcontrol.com/master/en/getting_started/ and attempted to build with Qt Creator.

Kim-zhi-jiang, 2022-01-25:
@mrpollo  I always use Qt to make build QGC, previously used qt-opensource-linux-x64-5.12.6

closed 2022-01-28 by Kim-zhi-jiang

rishi1906, 2022-02-06:
@Kim-zhi-jiang I am facing the same issue, did you get the solution?

hamishwillee, 2022-04-21:
The docs are wrong "by definition" - because you should be able to build by following the instructions.

The problem here is that Airmap is built by default on Linux, but there are no instructions for adding the airmap SDK. So either we need to add those instructions, or add instructions on how to disable the airmap build.

I will do the later, because it is easy - you create a file in the root called **user_config.pri** with the content `DEFINES += DISABLE_AIRMAP`

But we still need to know how to get airmap dependencies - see #10260

Kim-zhi-jiang, 2022-06-13:
echo "CONFIG = WarningsAsErrors" >> /home/xxx/qgroundcontrol/user_config.pri
echo "DEFINES = DISABLE_AIRMAP" >> /home/xxx/qgroundcontrol/user_config.pri