mavlink/qgroundcontrol #10140: Use upstream gst-plugins-good

acxz, 2022-01-28:
Tell us a bit about the feature:
- What problem does it solve?
  Using the upstream https://github.com/GStreamer/gst-plugins-good allows us to track changes much easily. In particular https://github.com/mavlink/qgroundcontrol/issues/9669 and the associated PR https://github.com/mavlink/gst-plugins-good/pull/1 would not have been needed had we used the upstream repo instead of our own fork.
- What does it do?
  Replace our fork of https://github.com/mavlink/gst-plugins-good with upstream https://github.com/GStreamer/gst-plugins-good
- What flight stacks must it work with? (All, PX4, ArduPilot)
  All
- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
  All
- Are there other systems that have this feature?
  N/A
- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)
  N/A

Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)

Following conversation from above mentioned issue and PR.

Looks like the only difference is this one commit: https://github.com/mavlink/gst-plugins-good/commit/9d782fad9dc0384ba86ecae64511c193f6149f93

Understanding what it does and if the capability has already been incorporated into upstream (if not then sending the appropriate PR upstream) will be needed to resolve this issue.

mrpollo, 2022-02-08:
thanks for taking the time to review @acxz we didn't have the time to upstream our changes to gst plugins, if you are willing to contribute the change to upstream gst-plugins-good we would appreciate it

acxz, 2022-02-08:
let me send it over to them and see what feedback i get

acxz, 2022-02-08:
This may be a shot in the dark and I dont know if they are active anymore. But if possible as the person who wrote the commit, @andrewvoznytsa can you comment on why the commit is needed in a more detailed manner than just the commit message?

I'd like to make an issue over at https://gitlab.freedesktop.org/gstreamer/gstreamer but since I don't really know the usecase for the commit, it would be great if you can comment and explain to us, so that I can make a proper detailed issue upstream.

acxz, 2022-02-09:
Got a response and it seems like the original issue has been resolved in version `1.20`. I can submit a PR to use the upstream repo as a submodule.

mrpollo, 2022-02-09:
Hey @patrickelectric you might have an input here

andrewvoznytsa, 2022-02-09:
@acxz I think you found your answers. If your target system has 1.19.2+ then it looks safe to drop that custom fork. For pre-1.19.2 it is pretty straight forward to adapt my patch.

patrickelectric, 2022-02-09:
> Got a response and it seems like the original issue has been resolved in version `1.19.2`. I can submit a PR to use the upstream repo as a submodule.

That sounds great!

smagellan, 2022-02-19:
gst-plugins-good moved to https://github.com/GStreamer/gstreamer/tree/1.20/subprojects/gst-plugins-good

acxz, 2022-02-25:
Thanks for that @smagellan i'm using the monorepo hosted at gitlab instead of the mirror tho. Thought it would be better to be closer to upstream.

closed 2024-05-31 by HTRamsey