mavlink/qgroundcontrol #10141: Memory overflow warning:  warning: ‘__builtin_strncpy’ writing 50 bytes into a region of size 24

AndKe, 2022-01-28:
This warning appears on every build on Linux. 
```
compiling moc_qtitem.cpp
linking staging/QGroundControl
In function ‘strncpy’,
    inlined from ‘_handleStatusText’ at /home/andre/prog/qgroundcontrol/src/Vehicle/Vehicle.cc:933:12:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:95:34: warning: ‘__builtin_strncpy’ writing 50 bytes into a region of size 24 overflows the destination [-Wstringop-overflow=]
   95 |   return __builtin___strncpy_chk (__dest, __src, __len,
      |                                  ^
/usr/include/x86_64-linux-gnu/bits/string_fortified.h: In member function ‘_handleStatusText’:
/home/andre/Qt/5.15.2/gcc_64/include/QtCore/qarraydata.h:129:8: note: destination object ‘<anonymous>’ of size 24
  129 | struct QTypedArrayData
      |        ^

```

acxz, 2022-02-26:
Can confirm:

```
/home/acxz/vcs/git/github/acxz/qgroundcontrol/src/Vehicle/Vehicle.cc: In member function ‘_handleStatusText’:
/home/acxz/vcs/git/github/acxz/qgroundcontrol/src/Vehicle/Vehicle.cc:933:12: warning: ‘strncpy’ writing 50 bytes into a region of size 24 overflows the destination [-Wstringop-overflow=]
  933 |     strncpy(b.data(), statustext.text, MAVLINK_MSG_STATUSTEXT_FIELD_TEXT_LEN);
      |            ^
/usr/include/qt/QtCore/qarraydata.h:129:8: note: destination object ‘<anonymous>’ of size 24
  129 | struct QTypedArrayData
      |        ^
```

acxz, 2022-08-06:
I can't seem to reproduce this warning on gcc 12.1.1 anymore.

acxz, 2022-08-07:
@AndKe can u close this issue if u can't repro it?

closed 2022-08-07 by AndKe