mavlink/qgroundcontrol #10144: Current master fail to download parameters or upload mission, / issue commands to connected SoloLink

AndKe, 2022-02-04:
## Expected Behavior
Expected it to "just work", as it did for years
Please have in mind that the SoloLink / IMX6 and some Solo-guts are used on many much bugger UAV's.

## Current Behavior
The green param download line stops short.
unable to set mode
unable to upload mission (times out)

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Connect wifi to a SoloLink  4.0  , with current ArduCopter
2.observe it "connect" instantly, then stop the green progress bar.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 21.10
- QGC Version: master
- QGC build: built from source
- Flight Controller: Cube2
- Autopilot (with version): AC 4.1.3

## Detailed Description
It seems it has trouble communicating, and at least issuing commands.
If I test using MAVProxy - then everything works as expected (proving that the computer/wifi/SoloLink/ArduCopter works as expected)

## Log Files and Screenshots
- What kind of data would you like?
- I could probably also share such a connection to the internet for troubleshooting purposes.

AndKe, 2022-02-06:
Correction: this happens when the ArduPilot speaks MAVlink 2.0  - if the serial protocol is set to MAVLink  1.0 - all works as expected.
This is most likely due to lack of support for MAVLink 2.0 issue within the IMX6 (just guessing)

AndKe, 2022-02-06:
Please consider to add an option to force MAVLink version to 1.0  - this is how MAVProxy can handle this issue.

mrpollo, 2022-02-08:
Hi @AndKe I regrettably don't have any hardware to debug this, but would consider reviewing and helping guide someone to submit a PR

closed 2026-10-04 by github-actions[bot]