mavlink/qgroundcontrol #10166: Segmentation fault on startup

kolesar-andras, 2022-02-15:
I am unable to start QGroundControl on a specific Ubuntu machine.

* Operating system: Ubuntu 20.04 LTS
* QGroundControl 4.x (any) latest downloaded today, SHA-1 of AppImage file is 2f13d215
* Autopuilot board: APM (not relevant because not connected yet)
* Autopilot firmware: ArduPlane 3.4.0 (latest for 8-bit APM, not relevant)
* Exact steps to reproduce the problem: starting application from terminal

```
$ ./QGroundControl.AppImage
Settings location "/home/kolesar/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(Hungarian, Latin, Hungary) ; Name "hu_HU" ; Preffered (used in maps):  "hu"
LocalizationLog: Qt lib localization for "hu_HU" is not present
LocalizationLog: Error loading source localization for "hu_HU"
LocalizationLog: Error loading json localization for "hu_HU"
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/kolesar/Dokumentumok/QGroundControl/Logs"
Map Cache in: "/home/kolesar/.cache/QGCMapCache300" / "qgcMapCache.db"
qml: QGCCorePlugin(0x55f5224f18b0) [1,2]
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
Segmentation fault (core dumped)
```

The same AppImage file works fine on another Ubuntu 20.04 LTS and 18.04 LTS systems.

zdanek, 2022-02-15:
@kolesar-andras try cleaning (or temp moving somewhere else) settings file `/home/kolesar/.config/QGroundControl.org/QGroundControl.ini`

kolesar-andras, 2022-02-15:
Thank you for your suggestion. Tried and I got the same error.

Application recreated the same .ini file there. The only difference is in `FlightMapZoom` setting, probably depends on monitor resolution used on first start.

An empty window pops up titled QGroundControl. There is a scroll bar in the middle, closes after a second with segfault.

kolesar-andras, 2022-02-17:
```
$ qmake --version
qmake: could not exec '/usr/lib/qt5/bin/qmake': No such file or directory
```

Installed `qmake` with `sudo apt install qt5-qmake-bin`, then checked again:

```
$ qmake --version
QMake version 3.1
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu
```

QGroundControl still gives segmentation fault on startup.

kolesar-andras, 2022-02-17:
Checked output of `qtdiag`. There are important differences.

Machine where QGroundControl runs fine:

```
LibGL Vendor: Mesa/X.org
Renderer: llvmpipe (LLVM 12.0.0, 256 bits)
Version: 3.1 Mesa 21.0.3
Shading language: 1.40
Format: Version: 3.1 Profile: 0 Swap behavior: 0 Buffer size (RGB): 8,8,8
Profile: None (QOpenGLFunctions_3_1), Core (QOpenGLFunctions_3_1_Core), Compatibility (QOpenGLFunctions_3_1_Compatibility)

Vulkan instance available
Supported instance extensions:
  VK_KHR_device_group_creation, version 1
[...]
```

Machine where QGroundControl exits with segmentation fault:

```
LibGL Vendor: Intel
Renderer: Mesa Intel(R) UHD Graphics 620 (WHL GT2)
Version: 4.6 (Compatibility Profile) Mesa 21.0.3
Shading language: 4.60
Format: Version: 4.6 Profile: 2 Swap behavior: 0 Buffer size (RGB): 8,8,8
Profile: None (QOpenGLFunctions_4_6)

Unable to create a Vulkan instance, error code is0
```

What should I check / install / uninstall / reinstall now?

kolesar-andras, 2022-02-17:
Problem may not be related to Vulkan. I have tried to start QGroundControl on a temporary installation of Ubuntu 21.04 on the same hardware. After installing required packages QGroundControl started fine. Checked `qtdiag` on this system and reported the same Vulkan errors.

Output on this working system ended with this section:

```
Qt Rendering Hardware Interface supported backends:
OpenGL (with default QSurfaceFormat):
  Min Texture Size: 1
  Max Texture Size: 16384
  Max Color Attachments: 8
  Frames in Flight: 1
  Uniform Buffer Alignment: 1
  Supported MSAA sample counts: 1,2,4,8,16
  Features:
    - MultisampleTexture
    v MultisampleRenderBuffer
    - DebugMarkers
    - Timestamps
    v Instancing
    - CustomInstanceStepRate
    v PrimitiveRestart
    v NonDynamicUniformBuffers
    v NonFourAlignedEffectiveIndexBufferOffset
    v NPOTTextureRepeat
    - RedOrAlpha8IsRed
    v ElementIndexUint
    v Compute
    v WideLines
    v VertexShaderPointSize
    v BaseVertex
    - BaseInstance
    v TriangleFanTopology
    v ReadBackNonUniformBuffer
    v ReadBackNonBaseMipLevel
  Texture formats: RGBA8 BGRA8 R8 R16 RED_OR_ALPHA8 RGBA16F RGBA32F R16F R32F D16 D32F BC1 BC2 BC3 ETC2_RGB8 ETC2_RGB8A1 ETC2_RGBA8
```

This is missing from the Ubuntu 20.04 LTS system where QGroundControl does not start. Another 20.04 LTS system where QGroundControl runs has Vulkan but no OpenGL section in `qtdiag` output.

I guess either Vulkan or OpenGL should be available. What can I do now?

lucasyu17, 2022-04-04:
Hi, I have met the same issue. Have you solved it?

zdanek, 2022-04-05:
@lucasyu17 please provide a crash dump like kolesar-andras above. Maybe it's something with OpenGL?

kolesar-andras, 2022-04-05:
I have also tried to run QGroundControl in docker container based on a virgin Ubuntu 20.04 using the window manager of the problematic Ubuntu 20.04 system. The same issue happened. I hope this helps a bit.

zdanek, 2022-04-05:
This might be something with your system setup. Docker abstracts only a thin layer from real system. So this might be for example something with your opengl.

kolesar-andras, 2022-04-05:
Appimage started successfully with `-platform vnc` command line parameter. Screen is small, colors are washed but works without segmentation fault. This experiment confirms problem is about graphics.

zdanek, 2022-04-05:
My intuition was correct :D Now we wait for dump from @lucasyu17 to see if he has also some OpenGL related logs. I have simple Intel built-in graphics card and no problems (and no performace as well ;) )

kolesar-andras, 2022-04-05:
Problem is not strictly hardware related. A fresh Ubuntu 20.04 installed on another partition on the same hardware runs QGgroundControl fine.

lucasyu17, 2022-04-05:
Here is a dump with QSG_INFO=1, I am running it on a ubuntu 18.04 machine with a nvidia graphic card. It worked before, but recently I re-installed my ubuntu system and it started to give this issue. Thanks!
`QSG_INFO=1 ./QGroundControl.AppImage 
Settings location "/home/lucas/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/lucas/Documents/QGroundControl/Logs"
VideoReceiverLog: Stop called on empty URI
Map Cache in: "/home/lucas/.cache/QGCMapCache300" / "qgcMapCache.db"
qt.scenegraph.general: threaded render loop
qt.scenegraph.general: Using sg animation driver
qt.scenegraph.general: Animation Driver: using vsync: 16.67 ms
qml: QGCCorePlugin(0x564984c87e90) []
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
qt.scenegraph.general: Using sg animation driver
qt.scenegraph.general: Animation Driver: using vsync: 16.67 ms
qt.scenegraph.general: opengl texture atlas dimensions: 4096x2048
qt.scenegraph.general: R/G/B/A Buffers:   8 8 8 0
qt.scenegraph.general: Depth Buffer:      24
qt.scenegraph.general: Stencil Buffer:    8
qt.scenegraph.general: Samples:           -1
qt.scenegraph.general: GL_VENDOR:         NVIDIA Corporation
qt.scenegraph.general: GL_RENDERER:       Quadro RTX 3000/PCIe/SSE2
qt.scenegraph.general: GL_VERSION:        4.6.0 NVIDIA 460.91.03
qt.scenegraph.general: GL_EXTENSIONS:    GL_AMD_multi_draw_indirect GL_AMD_seamless_cubemap_per_texture GL_AMD_vertex_shader_layer GL_AMD_vertex_shader_viewport_index GL_ARB_ES2_compatibility GL_ARB_ES3_1_compatibility GL_ARB_ES3_2_compatibility GL_ARB_ES3_compatibility GL_ARB_arrays_of_arrays GL_ARB_base_instance GL_ARB_bindless_texture GL_ARB_blend_func_extended GL_ARB_buffer_storage GL_ARB_cl_event GL_ARB_clear_buffer_object GL_ARB_clear_texture GL_ARB_clip_control GL_ARB_color_buffer_float GL_ARB_compatibility GL_ARB_compressed_texture_pixel_storage GL_ARB_compute_shader GL_ARB_compute_variable_group_size GL_ARB_conditional_render_inverted GL_ARB_conservative_depth GL_ARB_copy_buffer GL_ARB_copy_image GL_ARB_cull_distance GL_ARB_debug_output GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_derivative_control GL_ARB_direct_state_access GL_ARB_draw_buffers GL_ARB_draw_buffers_blend GL_ARB_draw_elements_base_vertex GL_ARB_draw_indirect GL_ARB_draw_instanced GL_ARB_enhanced_layouts GL_ARB_explicit_attrib_location GL_ARB_explicit_uniform_location GL_ARB_fragment_coord_conventions GL_ARB_fragment_layer_viewport GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_fragment_shader GL_ARB_fragment_shader_interlock GL_ARB_framebuffer_no_attachments GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_geometry_shader4 GL_ARB_get_program_binary GL_ARB_get_texture_sub_image GL_ARB_gl_spirv GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_gpu_shader_int64 GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_imaging GL_ARB_indirect_parameters GL_ARB_instanced_arrays GL_ARB_internalformat_query GL_ARB_internalformat_query2 GL_ARB_invalidate_subdata GL_ARB_map_buffer_alignment GL_ARB_map_buffer_range GL_ARB_multi_bind GL_ARB_multi_draw_indirect GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_occlusion_query2 GL_ARB_parallel_shader_compile GL_ARB_pipeline_statistics_query GL_ARB_pixel_buffer_object GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_polygon_offset_clamp GL_ARB_post_depth_coverage GL_ARB_program_interface_query GL_ARB_provoking_vertex GL_ARB_query_buffer_object GL_ARB_robust_buffer_access_behavior GL_ARB_robustness GL_ARB_sample_locations GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_seamless_cubemap_per_texture GL_ARB_separate_shader_objects GL_ARB_shader_atomic_counter_ops GL_ARB_shader_atomic_counters GL_ARB_shader_ballot GL_ARB_shader_bit_encoding GL_ARB_shader_clock GL_ARB_shader_draw_parameters GL_ARB_shader_group_vote GL_ARB_shader_image_load_store GL_ARB_shader_image_size GL_ARB_shader_objects GL_ARB_shader_precision GL_ARB_shader_storage_buffer_object GL_ARB_shader_subroutine GL_ARB_shader_texture_image_samples GL_ARB_shader_texture_lod GL_ARB_shader_viewport_layer_array GL_ARB_shading_language_100 GL_ARB_shading_language_420pack GL_ARB_shading_language_include GL_ARB_shading_language_packing GL_ARB_shadow GL_ARB_sparse_buffer GL_ARB_sparse_texture GL_ARB_sparse_texture2 GL_ARB_sparse_texture_clamp GL_ARB_spirv_extensions GL_ARB_stencil_texturing GL_ARB_sync GL_ARB_tessellation_shader GL_ARB_texture_barrier GL_ARB_texture_border_clamp GL_ARB_texture_buffer_object GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_buffer_range GL_ARB_texture_compression GL_ARB_texture_compression_bptc GL_ARB_texture_compression_rgtc GL_ARB_texture_cube_map GL_ARB_texture_cube_map_array GL_ARB_texture_env_add GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_filter_anisotropic GL_ARB_texture_filter_minmax GL_ARB_texture_float GL_ARB_texture_gather GL_ARB_texture_mirror_clamp_to_edge GL_ARB_texture_mirrored_repeat GL_ARB_texture_multi [... cut ...] NV_vertex_program2_option GL_NV_vertex_program3 GL_NV_viewport_array2 GL_NV_viewport_swizzle GL_OVR_multiview GL_OVR_multiview2 GL_S3_s3tc GL_SGIS_generate_mipmap GL_SGIS_texture_lod GL_SGIX_depth_texture GL_SGIX_shadow GL_SUN_slice_accum
qt.scenegraph.general: Max Texture Size: 32768
qt.scenegraph.general: Debug context:    false
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
QGCPositionManager error 0
"v4.2.0"`

zdanek, 2022-04-06:
I'm not fluent in reading this. I see some NVidia stuff.  I have ordinary intel video card so I cannot check it. But I think you @kolesar-andras and @lucasyu17 can cooperate to check various drivers / libraries regarding OpenGL and graphics. I bet it's the problem.

kolesar-andras, 2022-04-06:
I'm going to compare faulty and working Ubuntu systems. Can you provider some help for debugging? Command line options?

kolesar-andras, 2022-05-31:
Upgraded to Ubuntu 21.04. QGroundControl starts fine.

zdanek, 2022-10-10:
@kolesar-andras can you please close this issue?

closed 2024-12-01 by HTRamsey