mavlink/qgroundcontrol #10167: Incorrect first connection after restarting Cube Orange or SITL

bender20008, 2022-02-15:
## Expected Behavior
After restarting the autopilot, the first connection should work correctly

## Current Behavior
After restarting SITL or cube orange and the first connection QGC to the SITL or Cube Orange, the autopilot version is not detected and there is no HELI tab in Vehicle Setup. 

## Steps to Reproduce:
1. Restart SITL or Cube Orange
2. Open QGC and connect to SITL or Cube Orange
3. Open Vehicle Setup 
4. No tab Heli

## System Information
- Operating System: [macOS Big Sur 11.5.1]
- QGC Version: [v4.2.0]
- QGC build: [stable]
- Autopilot (with version): [Ardupilot Copter 4.0.7]

## Detailed Description
After restarting SITL or Cube Orange and connecting a ground station to it, an incorrect connection occurs. There is no Heli tab in the Vehicle Setup window. The problem is solved by restarting the QGC app. When reconnecting the QGC, the autopilot version is determined and the Heli tab is visible (Watch a video). 
For example, in stable v3.5.6, when you first connect, you can hear how QGC says "Communication lost" and immediately "Communication regained". There somehow the first incorrect connection is broken and a new one is created. 

## Log Files and Screenshots
https://user-images.githubusercontent.com/96539987/154066216-756fea91-d858-4dd7-9ccd-efac40a5a7b4.mp4

First bad connection:
[image]

Connection after restart QGC:
[image]

bender20008, 2022-02-24:
Does anyone have any ideas?

manavgandhi17, 2022-03-09:
I have found that you have to hit the "Disconnect" button that pops up before reconnecting the autopilot/SITL.

bender20008, 2022-03-09:
The problem is that the first time you connect to the autopilot (after rebooting the autopilot), a "_waitForMavlinkMessage: collision" occurs and the Heli tab is not displayed. After restarting the GCS, the problem goes away.

closed 2024-12-01 by HTRamsey