mavlink/qgroundcontrol #10172: ADSB use server name not IP address

timtuxworth, 2022-02-16:
Tell us a bit about the feature:
- What problem does it solve?
Who knows IP addresses these days? 
- What does it do?
Allow the ADSB server to be addressed by name
- What flight stacks must it work with? (All, PX4, ArduPilot)
All
- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
All
- Are there other systems that have this feature?
Pretty much the whole internet
- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)
adsb/dns

Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)
If a server changes the IP address of the dump1090 service could change. This might happen if a DHCP lease expires and a new IP is allocated, or if a server switches networks (e.g. if there is a failover network set up). If a domain name was used this would work dynamically. As it is, it will require manual intervention any time the server IP changes.

booo, 2022-02-18:
The `connectToHost` function that we use internally supports hostnames too. Did you try setting the value to a hostname? 

Maybe its just a matter of changing the description in the GUI.

timtuxworth, 2022-02-18:
> The `connectToHost` function that we use internally supports hostnames too. Did you try setting the value to a hostname?
> 
> Maybe its just a matter of changing the description in the GUI.

Yes I tried, the UI threw an error.

booo, 2022-02-18:
> > The `connectToHost` function that we use internally supports hostnames too. Did you try setting the value to a hostname?
> > Maybe its just a matter of changing the description in the GUI.
> 
> Yes I tried, the UI threw an error. Maybe its just a question of changing the validation on the UI?

Changing the value to "localhost" did not throw an error her. Can you post the error?

timtuxworth, 2022-02-18:
I was using raspberrypi.local and (I think) the RadarBox was down when I tried adding it. In fact the error wasn't thrown till I restarted QGC. I guess this means this Request is not required. 
[image]

closed 2022-02-18 by timtuxworth

booo, 2022-02-18:
I'm not sure if hostnames work. If you tested successfully with active host it think we should change the label of the option.

zdanek, 2022-02-19:
I've checked it and it works. I used my flightradar24 box host-name. 
I think it's important that no matter which addres will be used it must be resolvable to an IP. So `ping` this name and if it works it will work in QGC as well.

[image]

timtuxworth, 2022-02-19:
I closed it. I think "Host address" is ok as the label @booo - it doesn't say "IP address" and if we changed it to say hostname, it might mislead folks into thinking IP address wasn't an option. Host Address can be read to mean hostname or IP address. A tool tip would probably be helpful, but that seems to be a more general UI issue, none of the fields seem to have tool tips.