mavlink/qgroundcontrol #10173: ADSB display a different icon for planes and helicopters

timtuxworth, 2022-02-16:
- What problem does it solve?
When using ADSB to determine if there are vehicles around to be avoided, it's helpful to know if the vehicle is a plane or helicopter as helicopters typically more unpredictable as to altitude and need to be monitored more carefully for this, whereas planes typically fly at constant altitude or at least ascend/descend in a fairly predictable way.
- What does it do?
Display a different icon on the map for a plane vs a helicopter
- What flight stacks must it work with? (All, PX4, ArduPilot)
All
- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
Anything that flys I guess 
- Are there other systems that have this feature?
most ADSB websites like radarbox or flightaware display this 
- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)
Not sure - if there is a way to get the data from the data feed it would be ideal

booo, 2022-11-02:
Do you know which part of the adsb message contains the information about the aircraft type?

zdanek, 2022-11-02:
> Do you know which part of the adsb message contains the information about the aircraft type?  

@booo none. I did thorough fix and update of ADSB code somewhere in early 2022 and dug into this topic.
To know aircraft type we must have hex id database to distinct a type. Probably we could find a free service with an API (like adsbexchange.com ) or a partnership with some adsb service provider like flightradar24 or such.
This would also require active internet connection otherwise falling back to standard plane icon.  

This needs to be discussed in the Community. If there's a need for such functionality I could add it - I can find ADSB data source and code it into QGC. I'd love to take it because I found ADSB topic very interesting.

 But my concern is whether there's a wider need from the User's perspective? Any votes for this?

zdanek, 2022-11-02:
@booo it seems that we discussed it already with @timtuxworth and I even made some research, see:
https://github.com/mavlink/qgroundcontrol/pull/10170#issuecomment-1041496392

so we're on the track. Still waiting on any votes on this functionality.

timtuxworth, 2022-11-02:
I'd really like to see vehicle type as it makes a lot of difference for avoidance if it's a helicopter vs a plane. Ideally I'd then like to be able to send the data to the autopilot over mavlink.

closed 2026-10-05 by github-actions[bot]