mavlink/qgroundcontrol #10176: Required package missing from documentation

kolesar-andras, 2022-02-17:
QGroundControl does not start on a virgin Ubuntu 21.04 machine.

* Operating system: Ubuntu 21.04
* QGroundControl 4.x (any) latest downloaded today, SHA-1 of AppImage file is 2f13d215
* Autopuilot board: APM (not relevant because not connected yet)
* Autopilot firmware: ArduPlane 3.4.0 (latest for 8-bit APM, not relevant)
* Exact steps to reproduce the problem: starting application from terminal

Given I have done all doumented steps:
```
sudo usermod -a -G dialout $USER
sudo apt-get remove modemmanager -y
sudo apt install gstreamer1.0-plugins-bad gstreamer1.0-libav gstreamer1.0-gl -y
```

When I start QGroundControl, it gives the following error:

```
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

Aborted (core dumped)
```

After installing Qt with `sudo apt install libqt5gui5` QGroundControl starts fine.

This requirement is not documented on [download](https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html) page.

booo, 2022-02-18:
Cloud you update the documentation for your setup?

https://github.com/mavlink/qgc-user-guide/edit/master/en%2Fgetting_started%2Fdownload_and_install.md

kolesar-andras, 2022-02-18:
Done: https://github.com/mavlink/qgc-user-guide/pull/323

DonLakeFlyer, 2022-02-20:
If you are using the AppImage there should be no additional requirements for installation of other packages. It should just work. This would indiate something is wrong with the AppImage not that the docs are wrong.

DonLakeFlyer, 2022-02-20:
The updates to the docs to include an install of gstreamer stuff is incorrect as well. That should have been fixed by fixing the AppImage creation. The point of the AppImage is that it should just work out of the box.

DonLakeFlyer, 2022-02-20:
The doc update is certainly good for now since it's better than nothing. The title of this should changed to AppImage not working correctly.

thopiekar, 2022-03-12:
I can imagine that the library inside the AppImage depends on some other library that is missing or in a different version outside the AppImage.
@kolesar-andras If you are interested you can test the flatpak I build every week based on master. Would be interested if we get the same there.
- https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/

closed 2024-11-09 by HTRamsey