mavlink/qgroundcontrol #10177: There is a problem with the Chinese translation（qgc_json_zh_CN.ts ）

drbparadise, 2022-02-21:
Content in file：
```
    <message>
      <extracomment>.QGC.MetaData.Facts[indoorPalette].enumStrings, </extracomment>
      <location filename="../src/Settings/App.SettingsGroup.json"/>
      <source>Indoor,Outdoor</source>
      <translation>室内，室外</translation>
    </message>
```
For enumstrings strings, English commas will be used for segmentation in the code. In the translation, the English comma is replaced by the Chinese comma, which makes the program unable to run normally.

Code
```
bool FactMetaData::_parseEnum(const QJsonObject& jsonObject, DefineMap_t defineMap, QStringList& rgDescriptions, QStringList& rgValues, QString& errorString)
{
    rgDescriptions.clear();
    rgValues.clear();
    errorString.clear();

    if (!jsonObject.contains(_enumStringsJsonKey)) {
        return true;
    }

    QString strings = jsonObject.value(_enumStringsJsonKey).toString();
    rgDescriptions = defineMap.value(strings, strings).split(",", Qt::SkipEmptyParts);

    QString values = jsonObject.value(_enumValuesJsonKey).toString();
    rgValues = defineMap.value(values, values).split(",", Qt::SkipEmptyParts);

    if (rgDescriptions.count() != rgValues.count()) {
        errorString = QStringLiteral("Enum strings/values count mismatch - strings:values %1:%2").arg(rgDescriptions.count()).arg(rgValues.count());
        return false;
    }

    return true;
}
```
Log：
```
QML debugging is enabled. Only use this in a safe environment.
Settings location "C:/Users/durb/AppData/Roaming/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(Chinese, Simplified Han, China) ; Name "zh_CN" ; Preffered (used in maps):  "zh-CN"
LocalizationLog: Qt lib localization for "zh_CN" is not present
"FactMetaData::createFromJsonObject _parseEnum for offlineEditingFirmwareClass failed. Enum strings/values count mismatch - strings:values 1:3"
"FactMetaData::createFromJsonObject _parseEnum for indoorPalette failed. Enum strings/values count mismatch - strings:values 1:2"
"FactMetaData::createFromJsonObject _parseEnum for followTarget failed. Enum strings/values count mismatch - strings:values 1:3"
"FactMetaData::createFromJsonObject _parseEnum for gridLines failed. Enum strings/values count mismatch - strings:values 1:2"
"FactMetaData::createFromJsonObject _parseEnum for recordingFormat failed. Enum strings/values count mismatch - strings:values 1:3"
"FactMetaData::createFromJsonObject _parseEnum for apmChibiOS failed. Enum strings/values count mismatch - strings:values 1:2"
"FactMetaData::createFromJsonObject _parseEnum for apmVehicleType failed. Enum strings/values count mismatch - strings:values 1:5"
"FactMetaData::createFromJsonObject _parseEnum for streamRateRawSensors failed. Enum strings/values count mismatch - strings:values 2:14"
"FactMetaData::createFromJsonObject _parseEnum for streamRateExtendedStatus failed. Enum strings/values count mismatch - strings:values 2:14"
"FactMetaData::createFromJsonObject _parseEnum for streamRateRCChannels failed. Enum strings/values count mismatch - strings:values 2:14"
"FactMetaData::createFromJsonObject _parseEnum for streamRatePosition failed. Enum strings/values count mismatch - strings:values 2:14"
"FactMetaData::createFromJsonObject _parseEnum for streamRateExtra1 failed. Enum strings/values count mismatch - strings:values 2:14"
"FactMetaData::createFromJsonObject _parseEnum for streamRateExtra2 failed. Enum strings/values count mismatch - strings:values 2:14"
"FactMetaData::createFromJsonObject _parseEnum for streamRateExtra3 failed. Enum strings/values count mismatch - strings:values 2:14"
":/json/APM-MavCmdInfoCommon.json" " enum strings/values count mismatch, label:'设置' enumStrings:'开启，关闭'"
":/json/MavCmdInfoCommon.json" "MAV_CMD_NAV_CONTINUE_AND_CHANGE_ALT enum strings/values count mismatch, label:'模式' enumStrings:'爬升、空档、下降'"
"FactMetaData::createFromJsonObject _parseEnum for lock failed. Enum strings/values count mismatch - strings:values 1:8"
"FactMetaData::createFromJsonObject _parseEnum for lock failed. Enum strings/values count mismatch - strings:values 1:8"
MAVLinkLogManagerLog: MAVLink logs directory: "C:/Users/durb/Documents/QGroundControl Daily/Logs"
Map Cache in: "C:/Users/durb/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db"
```

Screenshot：
[image]

mrpollo, 2022-02-21:
would this be fixed if we replaced the Chinese comma? @drbparadise

drbparadise, 2022-02-22:
> would this be fixed if we replaced the Chinese comma? @drbparadise

Yes.  And also we should remind translators of all languages. Remind them to check all translations of enumstring types and make sure to use English comma as separator.

zdanek, 2022-02-22:
I think that there should be some kind of check. Maybe a python script that should be run at every translation change/update.
Maybe git hook or something. Asking people to remember about this is useless.

closed 2024-12-01 by HTRamsey