mavlink/qgroundcontrol #10178: QGC Crash When Saving Flight Mode - Rev1 Durandal

vincentpoont2, 2022-02-21:
## Expected Behavior
Flight mode should save, QGC should not crash.

## Current Behavior
QGroundControl Crash with Rev1 Durandal (Support added here https://github.com/PX4/PX4-Autopilot/pull/18628) when saving flight mode settings

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.) Using Rev 1 Durandal 
2.) QGC (Tried Daily & 4.1.4)
3.) Master PX4 FW
4.) When saving flight mode settings (Video: https://user-images.githubusercontent.com/46874772/154998452-af48217e-e0a6-4cf7-9d06-4f806723aa80.mp4)

## System Information
When posting bug reports, include the following information
- Operating System: Window 10
- QGC Version:  QGC Daily & 4.1.4)
- QGC build: daily, stable 
- Flight Controller: Durandal Rev1
- Autopilot (with version): Master

vincentpoont2, 2022-02-21:
Any idea? @mrpollo

mrpollo, 2022-02-21:
Has this always been an issue on Durandal? if not in which PX4 version did this start?

vincentpoont2, 2022-02-22:
Reproduced on both Pixhawk 4 & Durandal with Master FW, does not happen with Stable PX4 1.12.3.

davids5, 2022-02-22:
I can verify this on PX4 master. QGC crashes on both linux and windows.
Linux:
[1561688.874184] QGroundControl[176238]: segfault at 28 ip 00007f1c0c009b10 sp 00007ffdb483f658 error 4 in libQt5Core.so.5[7f1c0bd27000+5e4000]
[1561688.874195] Code: 66 2e 0f 1f 84 00 00 00 00 00 f6 47 0b 40 48 89 f8 75 07 f3 c3 0f 1f 44 00 00 48 8b 07 48 8b 00 c3 90 0f 1f 84 00 00 00 00 00 <8b> 4e 08 8b 47 08 4c 8b 06 81 e1 ff ff ff 3f 25 00 00 00 c0 09 c8

davids5, 2022-02-22:
LinkManagerVerboseLog: productIdentifier:  0
ParameterManagerLog: "V:1 C:1" Update parameter (_waitingParamTimeoutTimer started) - compId:name:rawValue 1 "RC_MAP_MODE_SW" QVariant(int, 5)
FactPanelControllerLog: Missing parameter: "1:RC_ASSIST_TH"
Segmentation fault (core dumped)

This appears to be related to https://github.com/PX4/PX4-Autopilot/commit/089c962d9208b3616ba505ab927fa399554035d4#diff-4190eb9f9b27f1693c2422281d1d26554c48256f150f8794315a8e6348e6d722L2102-L2229

mrpollo, 2022-02-23:
I spoke with @dagar this morning on the PX4 Developer Call (Feb 23), here's my notes:

PX4 v1.13 deprecated a few PARAMS and the multi-channel flight mode feature, we need to update QGC in a few areas:
* Remove setting multi-channel flight mode
* Do a quick pass through the source code for PX4 deprecated params (see links below)

https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/rc_update/params_deprecated.c

https://github.com/PX4/PX4-Autopilot/blob/5b5d4281891d671f82c346258dde1a555d0f303d/src/modules/rc_update/rc_update.cpp#L157-L202

https://github.com/PX4/PX4-Autopilot/blob/5b5d4281891d671f82c346258dde1a555d0f303d/src/modules/mavlink/mavlink_parameters.cpp#L282-L308

I think I'm going to open a new issue to track these tasks

closed 2022-03-02 by dagar

mrpollo, 2022-03-02:
@vincentpoont2 please let me know if you have any questions, we are pushing a new QGC release as soon as possible, but might take a few days, meanwhile please suggest the user who reported this issue to use the single-channel feature

mrpollo, 2022-03-02:
FYI I also removed this from the docs https://github.com/mavlink/qgc-user-guide/pull/326