mavlink/qgroundcontrol #10180: Comm Links view doesn't show

mrpollo, 2022-02-22:
Using the daily version of QGC I get a blank page, and running with logging turned on shows me this:

```
qrc:/qml/LinkSettings.qml:140:13: Type MockLinkOptionsDlg unavailable
qrc:/qml/QGroundControl/Controls/MockLinkOptionsDlg.qml: No such file or directory
```

[image]

mrpollo, 2022-02-22:
Seems to be related to 670a12b361 @DonLakeFlyer

DonLakeFlyer, 2022-02-22:
Yeah, most likely. I'll take a look.

highfreq, 2022-03-09:
Hello, just compiled from master last sunday and i can't see comm links too, is there a solution to this?
Running it on Android 12.

DonLakeFlyer, 2022-03-09:
Sorry lost track of this. I'll try to get to it today some time.

highfreq, 2022-03-09:
Actually i pulled the repo and recompiled and now it is there

DonLakeFlyer, 2022-03-09:
I verified it to be working fine on Ubuntu as well

closed 2022-03-09 by DonLakeFlyer