mavlink/qgroundcontrol #10189: QGC obstacle display errorneous orientation

hmloww, 2022-02-26:
## Expected Behavior
When obstacle overlay is enabled, the obstacles are overlaid on the UAV icon according to the obstacles specified in the OBSTACLE_DISTANCE message (https://mavlink.io/en/messages/common.html#OBSTACLE_DISTANCE) .

## Current Behavior
Obstacles are overlaid on the UAV icon but seems to be offset +90 degrees. (Obstacles prevention sensors on the aircraft: Stereo vision cam **pointing forward** with FOV around 60 degrees)
PX4 is behaving correctly preventing movement into obstacles straight ahead.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Start the system with a companion SBC send the OBSTACLE_DISTANCE message to both the PX4 and QGC.
2. Inspect the obstacles displayed

## System Information
- Operating System: Win 10
- QGC Version: v4.2.0
- QGC build: Stable
- Flight Controller: ModalAI Flight Core
- Autopilot (with version): PX4 1.11
- SBC: ModalAI VOXL
- Obstacles prevention sensors: Stereo vision cam pointing forward with FOV around 60 degrees

## Log Files and Screenshots
[image]
Obstacles prevention sensors on the aircraft: Stereo vision cam **pointing forward** with FOV around 60 degrees; PX4 is behaving correctly preventing movement into obstacles straight ahead.

dirksavage88, 2022-09-29:
@hmloww was this issue ever resolved? I am seeing this with QGC daily (linux) as well, I also am publishing directly to obstacle_distance map, I am not sure why it is only showing distance readouts for that offset angle and not the other obstacle sectors (maybe a sensor fov issue?) The distance readouts in that one sector also vary enough for me to believe it's not showing the distance readouts for just that sector in the obstacle distance maps (but other sectors as well, as I have 320 degree coverage).
[image]

DonLakeFlyer, 2023-02-18:
Is there some way to test this with SITL?

dirksavage88, 2023-02-26:
@bperseghetti do you know if representing mavlink obstacle_distance msgs coming from a gazebo rangefinder in SITL is feasible? What are the steps to get this working?

DonLakeFlyer, 2023-02-26:
If you can get it running in SITL then first step is to create a set of repro steps which show the problem happening there. I thought maybe the typhoon model has obstacle detection.

Simon-Pokorny, 2024-01-17:
You can reproduce the error by sending this Obstacle Distance message to /fmu/in/obstacke_distance. 

```
---
timestamp: 
frame: 12 
sensor_type: 0
distances:
- 100
- 200
- 300
- 400
- 500
- 600
- 700
- 800
- 900
- 1000
- 1100
- 1200
- 1300
- 1400
- 1500
- 1600
- 1700
- 1800
- 1900
- 2000
- 2100
- 2200
- 2300
- 2400
- 2500
- 2600
- 2700
- 2800
- 2900
- 3000
- 3100
- 3200
- 3300
- 3400
- 3500
- 3600
- 3700
- 3800
- 3900
- 4000
- 4100
- 4200
- 4300
- 4400
- 4500
- 4600
- 4700
- 4800
- 4900
- 5000
- 5100
- 5200
- 5300
- 5400
- 5500
- 5600
- 5700
- 5800
- 5900
- 6000
- 6100
- 6200
- 6300
- 6400
- 6500
- 6600
- 6700
- 6800
- 6900
- 7000
- 7100
- 7200
increment: 5.0
min_distance: 0
max_distance: 10000
angle_offset: 0.0
---
```
The angle offset is set to zero and in the screenshot below you can see that the origin is pointing to the right but it is supposed to point to the front. The behavior of px4 works like expected but the visualisation in qgc is wrong. 
[image]
The visualisation is wrong by 90 degree no matter which frame you choose.
I tried with the MAV_FRAME_GLOBAL which is north aligned and according to the [documentation](https://mavlink.io/en/messages/common.html#OBSTACLE_DISTANCE) the distance of obstacles around the vehicle with index 0 is corresponding to north + angle_offset.

Simon-Pokorny, 2024-03-22:
@DonLakeFlyer can you verify that the visualisation is off by 90° ?
If you verify that I could try to find a fix

arpitpara, 2024-03-27:
Hi 
I am also trying to send the same message from a ros2 node from my companion computer with a Lidar sensor. Everything seems to be fine but I can't see any indication in QGC that I am recieving the message.

Claudio-Chies, 2024-07-12:
@DonLakeFlyer There is now a PX4 SITL setup to reproduce the problem.
https://github.com/PX4/PX4-Autopilot/pull/22418
Steps to reproduce:

1. Get PX4 Branch above
2. Run SITL
`make px4_sitl gz_x500_lidar`
3. Drop a Box in Gazebo in front of the Drone
[image]

arpitpara, 2024-07-12:
Yeah. Sorry for not updating. This flow works for me. I have tried the setup in SITL and made it work in actual too. You need to do some Hit and trial while implementation in real. PX4 Collision Prevention is only supported in manual mode with one of 4 options available while setting Collision Prevention.

Claudio-Chies, 2024-12-05:
Apparently there's an open PR for it:
https://github.com/mavlink/qgroundcontrol/pull/10879#issuecomment-2516652426

closed 2026-10-04 by github-actions[bot]