mavlink/qgroundcontrol #10194: PX4 Autotuning UI not displayed for Fixed Wing

hamishwillee, 2022-03-03:
Autotuning is supported on PX4 for MC, FW and VTOL vehicles, but the UI is not shown in QGC for fixed wing (you can still trigger it using `FW_AT_START` but that is undesirable.

The UI should be updated to allow the feature to be used.

Jaeyoung-Lim, 2022-08-27:
Greatly appreciate @hamishwillee that this was clear from the docs.

Would you know who I would need to talk to in order to enable Auto tuning for fixed wings?
At least it would be great if the plots are shown so it can be used for manually hand tuning

bkueng, 2022-08-31:
> I was struggling to understand why the PID tuning UI would not appear for fixed wing vehicles

It's all ready, I just never got a definitive list of plots from @RomanBapst.

> Is this a bug? or intentionally blocked to prevent it from being used?

I think it was just never added to the UI.

Jaeyoung-Lim, 2022-08-31:
@bkueng Thanks! @RomanBapst Could we move this forward?

It is painful(or impossible) to tune the vehicle with mavlink inspector since the `VEHICLE_ATTITUDE` message is in quaternions and `VEHICLE_ATTITUDE_SETPOINT` is in euler angles

hamishwillee, 2022-09-01:
> I think it was just never added to the UI.

Yes, this was an omission. Would be good if @RomanBapst could provide the required info!!!

Jaeyoung-Lim, 2022-09-01:
@hamishwillee I ended up fixing it myself! https://github.com/mavlink/qgroundcontrol/pull/10384

closed 2022-09-09 by bkueng