mavlink/qgroundcontrol #10199: Servo Output Functions Page (w/ Control?)

manavgandhi17, 2022-03-07:
Would be nice to have a page where you can set the servo outputs to different functions. That way when I am setting up my planes in ArduPlane I don’t have to mess with the parameter list for basic setup. Don’t know if this should be its own page - or merged into the Motors page and renamed to “Actuators” or something like that.

AhmedElsafy, 2022-03-09:
Have you tried flashing the version 1.13 beta developer, you will find a page in QGC with actuator allocation that you can assign for each group of actuators the mode they work in

manavgandhi17, 2022-03-09:
I have tested on both the current stable and latest daily build. What do you mean v1.13? I am running ArduPlane.

AhmedElsafy, 2022-03-09:
Sorry my bad, I was more referring to PX4, it has that feature to control every output.