mavlink/qgroundcontrol #10200: Video fails when playing UDP/MPEGTS video stream

undefineV, 2022-03-09:
I tried to play the UDP video stream with the V4.0.8 version, it can be played normally, but with the latest version to play the same UDP video stream, I found that the video playback failed

## Expected Behavior
I would expect the latest version to also successfully play UDP video streams

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. I use VLC player to stream RTSP video stream to UDP video stream
2. Open the QGC ground station and select MPEG-TS (h.264) in the video stream source

## System Information
When posting bug reports, include the following information
- Operating System: [Windows 10]
- QGC Version: [V4.0.8 and V4.2.0]

## Log Files and Screenshots
This is the log information output by the application: 
setCurrentPlanViewSeqNum
QString::arg: Argument missing: "项目 1" , 0
setCurrentPlanViewSeqNum
VideoReceiverLog: Streaming started "mpegts://0.0.0.0:2000"
VideoReceiverLog: Starting decoding "mpegts://0.0.0.0:2000"
VideoReceiverLog: Decoding started "mpegts://0.0.0.0:2000"
"v4.2.0"
"FactMetaData::createFromJsonObject _parseEnum for lock failed. Enum strings/values count mismatch - strings:values 1:8"
"FactMetaData::createFromJsonObject _parseEnum for lock failed. Enum strings/values count mismatch - strings:values 1:8"
"FactMetaData::createFromJsonObject _parseEnum for lock failed. Enum strings/values count mismatch - strings:values 1:8"
"FactMetaData::createFromJsonObject _parseEnum for lock failed. Enum strings/values count mismatch - strings:values 1:8"
VideoManagerLog: No sense to restart video streaming, skipped 0
VideoManagerLog: Restart video streaming 1
QString::arg: Argument missing: "项目 1" , 0
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
ParameterManagerLog: Attemping load from cache
ParameterManagerLog: Parameters cache match failed C:/Users/Lyy/AppData/Roaming/QGroundControl.org/ParamCache/1_1.v2
QString::arg: Argument missing: "项目 1" , 0
QString::arg: Argument missing: "项目 1" , 1
QString::arg: Argument missing: "项目 1" , 2
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum

undefineV, 2022-03-09:
I tried to modify the code gst_element_factory_make("decodebin3", nullptr) in the file GstVideoReceiver.cc to gst_element_factory_make("decodebin", nullptr), then the video plays normally.

goasChris, 2022-04-25:
I would just like to add my voice to the mix as well.
Trying to use "TCP-MPEG2 video Stream" on later versions will not work. Logs are filled with the following:

```
[E] at ../libs/gst/base/gstbasesrc.c:3127 - "<source> error: Internal data stream error."
[E] at ../libs/gst/base/gstbasesrc.c:3127 - "<source> error: streaming stopped, reason not-linked (-1)"
[!] at C:\projects\qgroundcontrol\src\VideoReceiver\GstVideoReceiver.cc:1305 - "GStreamer error: Internal data stream error."
```
Repeating endlessly.

The latest working version I found was **4.0.11**.

The feed is coming from the latest gStreamer on ARM, its an mpegTS muxed stream provided as a tcpserversink.
Seems to be working fine in ffmpeg, OBS, VLC etc.

I have a feeling that it could be due to timing out too quickly when waiting for the next IDR frame, but this is speculation from my side. It opens up a large number of connections, and seems to exit them fairly quickly.

Video info
```
Input #0, mpegts, from 'tcp://192.168.218.3:3001':
  Duration: N/A, start: 19180.533322, bitrate: N/A
  Program 1
  Stream #0:0[0x41]: Video: h264 (Main) (HDMV / 0x564D4448), yuv420p(progressive), 768x432, 15 fps, 15 tbr, 90k tbn, 30 tbc
  ```
Example of gStreamer (server side), working fine on the older versions, as well as ffmpeg, VLC etc.
```
! v4l2h264enc extra-controls="controls,video_bitrate_mode=0,video_bitrate=200000,repeat_sequence_header=1,h264_i_frame_period=30" \
! 'video/x-h264,profile=main,level=(string)4.2' \
! mpegtsmux ! queue \
! tcpserversink host=$IP port=3001
```
I don't really have a dev environment set up at the moment to bisect, but hopefully this narrows down the correct version a bit.
Thank you in advance!

fredowski, 2022-09-08:
I generated a TCP Stream with 

`gst-launch-1.0 videotestsrc is-live=true ! 'video/x-raw, width=640, height=480, format=I420' ! x264enc ! mpegtsmux ! queue ! tcpserversink host=127.0.0.1 port=5600`

and that works with the build from PR #10396. I set the source in QGC to "TCP MPEG2 Video Stream" and the url to "127.0.0.1:5600".

closed 2024-05-31 by HTRamsey