mavlink/qgroundcontrol #10202: Beginner question

highfreq, 2022-03-09:
I just finished setting up QT5 environment on Linux and it finally compiles and runs.

I cloned the repository on sunday as per instruction (i guess default clone is master), now i would like to pull all the modifications made to master in the last 2 days, i went into qgroundcontrol directory and did a git fetch, but it looks like it didn't update anything.
Anybody would care to let me know what to do when i want to pull latest code, thanks in advance.

regards,

Corrado

zdanek, 2022-03-09:
@highfreq do you have your own Fork? Or pulling out of official repo?

zdanek, 2022-03-09:
If forked, you need to pull from offcial QGC repo. See some tips 
https://stackoverflow.com/questions/24815952/git-pull-from-another-repository

highfreq, 2022-03-09:
I did the following on sunday:

git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git

Thank today i went into qgroundcontrol dir and did git fetch

maybe i need to fetch submodules too? If so how?

zdanek, 2022-03-09:
See the diff between fetch and pull.
https://www.freecodecamp.org/news/git-fetch-vs-pull/#:~:text=git%20fetch%20is%20the%20command,changes%20from%20the%20remote%20repository.

Please try 
git pull

zdanek, 2022-03-09:
If yo want to do some more with development I suggest you some basic course of git. 
Good luck :)

highfreq, 2022-03-09:
Did a git pull and told me "already updated", read differences between pull and fetch, thanks for the link.
Do you know if i have to pull submodules too?

zdanek, 2022-03-09:
updating submodules in git, if any
git submodule update --recursive

but just check the history with
git log

you should see some latest commit. If so. Just go on with your work with the code :)
I heartly re

highfreq, 2022-03-09:
Yes see them with git log, will take a look at git bash prompt, thanks.

Is it correct to assume that git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git clones the master?

booo, 2022-04-17:
I guess what you are looking for is `git pull` to incorporate changes from a remote location. Git fetch only fetches updates from remote locations. Please take a look at the manpages or git docu: https://git-scm.com/book/en/v2

Closing this ticket because it's not related to QGroundControl bugs or features.

closed 2022-04-17 by booo