mavlink/qgroundcontrol #10205: run failed in Ubuntu20.04-arm64

mengchaoheng, 2022-03-10:
## Current Behavior
bash: ./QGroundControl.AppImage: cannot execute binary file: Exec format error

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1)  chmod +x ./QGroundControl.AppImage
2)  ./QGroundControl.AppImage  (or double click)

mengchaoheng, 2022-03-10:
how to install QGC in the focal-desktop-arm64 ubuntu?

mengchaoheng, 2022-03-11:
when I buil it on arm-baed ubuntu20.04
`
02:22:37: Running steps for project qgroundcontrol...
02:22:37: Configuration unchanged, skipping qmake step.
02:22:37: Starting: "/usr/bin/make" -j8
/usr/lib/qt5/bin/qmake -o Makefile /home/parallels/qgroundcontrol/qgroundcontrol.pro -spec linux-g++ CONFIG+=debug CONFIG+=qml_debug
Project MESSAGE: Qt version 5.12.8
Project MESSAGE: Linux build
Project MESSAGE: Daily Build
Project MESSAGE: QGroundControl v4.1.6
Project MESSAGE: Debug flavor
Project MESSAGE: Using Default QtLocation headers
Project MESSAGE: Skipping support for Pairing
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Looking for Airmap in folder "/home/parallels/qgroundcontrol/libs/airmapd", variant: "Qt.5.12"
Project MESSAGE: Taisync disabled
Project MESSAGE: Microhard disabled
Project MESSAGE: Skipping support for video streaming (manual override from command line)
Project ERROR: Unknown module(s) in QT: location-private positioning-private
make: *** [Makefile:2119: Makefile] Error 3
02:22:38: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Desktop)
When executing step "Make"
02:22:38: Elapsed time: 00:00.
`

`
parallels@ubuntu-linux-20-04-desktop:~/qgroundcontrol$ git status 
On branch Stable_V4.1
Your branch is up to date with 'origin/Stable_V4.1'.

parallels@ubuntu-linux-20-04-desktop:~/qgroundcontrol$ qmake -v
QMake version 3.1
Using Qt version 5.12.8 in /usr/lib/aarch64-linux-gnu
`

mengchaoheng, 2022-03-11:
@dogmaphobic

mengchaoheng, 2022-03-11:
[image]

mengchaoheng, 2022-03-11:
@patrickelectric

mengchaoheng, 2022-03-11:
@DonLakeFlyer

DonLakeFlyer, 2022-03-11:
You have not followed the dev build instructions in the docs.

mengchaoheng, 2022-03-11:
> You have not followed the dev build instructions in the docs.

I should, but I can't. Because it's a arm-based ubuntu, qt online install didn't work, so I have to install it manually.

thopiekar, 2022-04-17:
@mengchaoheng You should be able to build it without airmap SDK. All the other dependencies are open.
If you like you can use my arm64 builds via flatpak: https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/

mengchaoheng, 2022-04-18:
> @mengchaoheng You should be able to build it without airmap SDK. All the other dependencies are open.
> 
> If you like you can use my arm64 builds via flatpak: https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/

In fact, I would like to know how you compiled and installed qt5 in arm ubuntu, thank you very much.

mengchaoheng, 2022-04-18:
> @mengchaoheng You should be able to build it without airmap SDK. All the other dependencies are open. If you like you can use my arm64 builds via flatpak: https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/

install failed! output is :
<details><summary>CLICK ME</summary>
<p>

```Console

parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$  sudo apt install flatpak
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libappstream-glib8 libostree-1-1
The following NEW packages will be installed:
  flatpak libappstream-glib8 libostree-1-1
0 upgraded, 3 newly installed, 0 to remove and 116 not upgraded.
Need to get 1,672 kB of archives.
After this operation, 8,103 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 https://mirrors.ustc.edu.cn/ubuntu-ports focal/main arm64 libappstream-glib8 arm64 0.7.16-1ubuntu1 [124 kB]
Get:2 http://ppa.launchpad.net/alexlarsson/flatpak/ubuntu focal/main arm64 libostree-1-1 arm64 2020.8-flatpak1~focal [275 kB]
Get:3 http://ppa.launchpad.net/alexlarsson/flatpak/ubuntu focal/main arm64 flatpak arm64 1.12.4-1flatpak1~20.04 [1,272 kB]
Fetched 1,672 kB in 7s (253 kB/s)                                                                              
Selecting previously unselected package libappstream-glib8:arm64.
(Reading database ... 195542 files and directories currently installed.)
Preparing to unpack .../libappstream-glib8_0.7.16-1ubuntu1_arm64.deb ...
Unpacking libappstream-glib8:arm64 (0.7.16-1ubuntu1) ...
Selecting previously unselected package libostree-1-1:arm64.
Preparing to unpack .../libostree-1-1_2020.8-flatpak1~focal_arm64.deb ...
Unpacking libostree-1-1:arm64 (2020.8-flatpak1~focal) ...
Selecting previously unselected package flatpak.
Preparing to unpack .../flatpak_1.12.4-1flatpak1~20.04_arm64.deb ...
Unpacking flatpak (1.12.4-1flatpak1~20.04) ...
Setting up libappstream-glib8:arm64 (0.7.16-1ubuntu1) ...
Setting up libostree-1-1:arm64 (2020.8-flatpak1~focal) ...
Setting up flatpak (1.12.4-1flatpak1~20.04) ...
Processing triggers for man-db (2.9.1-1) ...
Processing triggers for dbus (1.12.16-2ubuntu2.1) ...
Processing triggers for libc-bin (2.31-0ubuntu9.7) ...
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ flatpak remote-add --if-not-exists thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/parallels/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ flatpak install thopiekar.eu org.mavlink.qgroundcontrol

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/parallels/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

Looking for matches…
error: The application org.mavlink.qgroundcontrol/aarch64/stable requires the runtime org.kde.Platform/aarch64/5.15 which was not found
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ sudo flatpak remote-add --if-not-exists thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ sudo flatpak install thopiekar.eu org.mavlink.qgroundcontrol
Looking for matches…
error: The application org.mavlink.qgroundcontrol/aarch64/stable requires the runtime org.kde.Platform/aarch64/5.15 which was not found
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ flatpak remote-add --if-not-exists thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo

[... 89 lines cut ...]
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ sudo flatpak remote-add --if-not-exists thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo

parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ 
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ sudo flatpak install thopiekar.eu org.mavlink.qgroundcontrol
Looking for matches…
error: The application org.mavlink.qgroundcontrol/aarch64/stable requires the runtime org.kde.Platform/aarch64/5.15 which was not found
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ flatpak update 

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/parallels/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

Looking for updates…
Nothing to do.
sparallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ 

```

</p>
</details>

mengchaoheng, 2022-04-18:
@thopiekar 

```
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ sudo flatpak install flathub org.kde.Platform
Looking for matches…
Similar refs found for ‘org.kde.Platform’ in remote ‘flathub’ (user):

   1) runtime/org.kde.Platform/aarch64/5.10
   2) runtime/org.kde.Platform/aarch64/5.11
   3) runtime/org.kde.Platform/aarch64/5.13
   4) runtime/org.kde.Platform/aarch64/5.12
   5) runtime/org.kde.Platform/aarch64/5.14
   6) runtime/org.kde.Platform/aarch64/5.15
   7) runtime/org.kde.Platform/aarch64/5.9
   8) runtime/org.kde.Platform/aarch64/5.15-21.08
   9) runtime/org.kde.Platform/aarch64/6.2

Which do you want to use (0 to abort)? [0-9]: 6
Skipping: org.kde.Platform/aarch64/5.15 is already installed
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ flatpak install thopiekar.eu org.mavlink.qgroundcontrol

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/parallels/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

Looking for matches…
^Cparallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ sudo flatpak install thopiekar.eu org.mavlink.qgroundcontrol
Looking for matches…
error: The application org.mavlink.qgroundcontrol/aarch64/stable requires the runtime org.kde.Platform/aarch64/5.15 which was not found
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ sudo flatpak install thopiekar.eu org.kde.Platform
Looking for matches…
error: Nothing matches org.kde.Platform in remote thopiekar.eu
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ 
```

thopiekar, 2022-04-18:
You need to run the following I guess:
`flatpak install flathub org.kde.Platform/aarch64/5.15`

And try to avoid `sudo` whenever possible. It ia not the medicine for every illness.

mengchaoheng, 2022-04-18:
> You need to run the following I guess: `flatpak install flathub org.kde.Platform/aarch64/5.15`
> 
> And try to avoid `sudo` whenever possible. It ia not the medicine for every illness.

and how to run qgc? I can't find it:
```
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/parallels/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ flatpak install flathub org.kde.Platform/aarch64/5.15

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/parallels/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

Looking for matches…
Info: org.kde.Platform//5.15 is end-of-life, with reason:
   We strongly recommend moving to the latest stable version of the Plaform and SDK

        ID                                                     Branch    Op    Remote     Download
 1. [✓] org.freedesktop.Platform.GL.default                    20.08     i     flathub     74.5 MB / 75.1 MB
 2. [✓] org.gtk.Gtk3theme.Yaru                                 3.22      i     flathub    136.7 kB / 191.4 kB
 3. [✓] org.kde.Platform.Locale                                5.15      i     flathub     17.9 kB / 341.5 MB
 4. [✓] org.kde.PlatformTheme.QGnomePlatform                   5.15      i     flathub      7.0 MB / 7.0 MB
 5. [✓] org.kde.PlatformTheme.QtSNI                            5.15      i     flathub      1.3 MB / 1.3 MB
 6. [✓] org.kde.WaylandDecoration.QGnomePlatform-decoration    5.15      i     flathub      3.2 MB / 7.5 MB
 7. [✓] org.kde.Platform                                       5.15      i     flathub    287.6 MB / 352.2 MB

Installation complete.
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ flatpak install flathub org.kde.Platform/aarch64/5.15

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/parallels/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

Looking for matches…
Skipping: org.kde.Platform/aarch64/5.15 is already installed
parallels@ubuntu-linux-20-04-desktop:~/PX4-Autopilot$ flatpak install thopiekar.eu org.mavlink.qgroundcontrol

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/parallels/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

Looking for matches…
Info: (pinned) org.kde.Platform//5.15 is end-of-life, with reason:
   We strongly recommend moving to the latest stable version of the Plaform and SDK

org.mavlink.qgroundcontrol permissions:
    ipc      network              pulseaudio           wayland      x11
    dri      file access [1]      dbus access [2]

    [1] home, xdg-config/kdeglobals:ro
    [2] com.canonical.AppMenu.Registrar

        ID                                 Branch         Op         Remote               Download
 1. [✓] org.mavlink.qgroundcontrol         stable         i          thopiekar.eu         27.9 MB / 28.4 MB

Installation complete.
```

mengchaoheng, 2022-04-18:
@thopiekar get it!!! let me summary as:
1.  sudo apt install flatpak
2. flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo
3. flatpak install flathub org.kde.Platform/aarch64/5.15
4. flatpak remote-add --if-not-exists thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo
5. flatpak install thopiekar.eu org.mavlink.qgroundcontrol

thopiekar, 2022-04-18:
So it runs? If so I'll add the flathub steps to my quickstart guide later.

mengchaoheng, 2022-04-18:
> So it runs? If so I'll add the flathub steps to my quickstart guide later.

yes, thank you very much!!!

mengchaoheng, 2022-04-19:
> So it runs? If so I'll add the flathub steps to my quickstart guide later.

New problem, I can't connect any hardware to the modified version of qgc, I tested v5 nano and pixhawk 4, they can connect to my qgc under macOS, but the modified version of qgc get the following console output：
```
[E) at :0 - "Qt: Session management error: None of the authentication protocols specified are supported"
[DI.at/run/build/qgroundcontrol/src/QGCApplication.cc:281 - "Settings location"/home/parallels/.var/pp/
org.mavlink.qgroundcontrol/config/QGroundControl.org/QGroundControl Daily.in'" Is writable?: true"
[D] at/run/build/qgroundcontrol/src/QGCLoggingCategory.cc:120 - "Filter rules
"*Log.debug-falselnGStreamerAPILog.debug-true\nqt.qml.connections=false™'
[D] at /run/build/qgroundcontrol/src/QGCApplication.cc:384-"System reported locale: QLocale(English, Latin, United States) ;
Name "en US" ; Preffered (used in maps): "en-US''
[ at :0 - "No text-to-speech plug-ins were found."
[D] at/run/build/agroundcontrol/sre/Vehicle/MAVLinkLogManager.cc:360-"MAVLink logs directory: "/home/parallels/
Documents/QGround Control Daily/Logs"'
[D]at/run/build/agroundcontrol/sr/QLocationPlugin/QGCMapEngine.cpp:164-"Map Cache in: "/home/parallels/.var/app/
org.mavlink.qgroundcontrol/cache/QGCMapCache300"/"agcMapCache.db''
[D]at/run/build/qgroundcontrol/sre/MissionManager/MissionController.cc:2370-"setCurrentPlanViewSegNum"
[D] at/run/build/qgroundcontrol/src/MissionManager/MissionController.cc:2370-"setCurrentPlanViewSeqNum"
[E] at/run/build/qgroundcontrol/src/VideoReceiver/GstVideoReceiver.cc:284- "Stop called on empty URI"
[E] at /run/build/agroundcontrol/src/VideoReceiver/GstVideoReceiver.cc:284-"Stop called on empty URI"

```

I successfully compiled and uploaded PX4 under my ubuntu20, I made sure my usb is connected to the computer and at /dev/ttyACM0.

mengchaoheng, 2022-04-19:
@thopiekar can't connect
[image]

mengchaoheng, 2022-04-19:
can you just add the stable branch to the thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo ? maybe daliy build have some bug!

mengchaoheng, 2022-04-19:
@thopiekar Can you teach me how to compile qgc on the ARM ubuntu? I don't know QT, I don't know how to set the compile option.

thopiekar, 2022-04-20:
Well, the reason you can't connect to serial devices is "normal". Flatpak apps run in sandboxes and giving the app permissions for accessing /dev/* might too open, so I kept it turned off.
However, you can install flatseal and allow QGroundControll to connect to all devices or run QGC via:
`flatpak run --device=all org.mavlink.qgroundcontrol`

Depending on the feedback from other people, I might grant the permission by default.

thopiekar, 2022-04-20:
> @thopiekar Can you teach me how to compile qgc on the ARM ubuntu? I don't know QT, I don't know how to set the compile option.

Don't have the time for this. Check out the general docs and dig into some scripts. Actually, it isn't too difficult.

retoc71586, 2022-06-07:
Hi @mengchaoheng did you manage to install qgroundcontrol or set up the whole px4 developer enviroment? I'm on ubuntu parallels on mac m1 too and I cant manage to install qgroundcontrol :(

mengchaoheng, 2022-06-07:
> Hi @mengchaoheng did you manage to install qgroundcontrol or set up the whole px4 developer enviroment? I'm on ubuntu parallels on mac m1 too and I cant manage to install qgroundcontrol :(

Yes, about qgc, see previous reply.

mengchaoheng, 2022-06-07:
> Hi @mengchaoheng did you manage to install qgroundcontrol or set up the whole px4 developer enviroment? I'm on ubuntu parallels on mac m1 too and I cant manage to install qgroundcontrol :(

https://github.com/mavlink/qgroundcontrol/issues/9856#issuecomment-1101268462

mengchaoheng, 2022-06-07:
> Hi @mengchaoheng did you manage to install qgroundcontrol or set up the whole px4 developer enviroment? I'm on ubuntu parallels on mac m1 too and I cant manage to install qgroundcontrol :(

Run qgc by 'flatpak run --device=all org.mavlink.qgroundcontrol' on the terminal.

mengchaoheng, 2022-08-30:
> 

sorry, after it work some day, it can't connect now. I have run qgc by flatpak run --device=all org.mavlink.qgroundcontrol. @thopiekar can you help me please!!!

mengchaoheng, 2022-08-30:
> 好吧，您无法连接到串行设备的原因是“正常”。Flatpak 应用程序在沙盒中运行，并且授予应用程序访问 /dev/* 的权限可能过于开放，因此我将其关闭。 但是，您可以安装 flatseal 并允许 QGroundControll 连接到所有设备或通过以下方式运行 QGC： `flatpak run --device=all org.mavlink.qgroundcontrol`
> 
> 根据其他人的反馈，我可能会默认授予权限。

I have install flatseal to setup but it didn't work. @thopiekar

gorghino, 2022-09-05:
We're facing the same problem @mengchaoheng got last Spring.
On a Nvidia NX (18.04):

```
drace@nx:~$ flatpak --user remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo
drace@nx:~$ flatpak install flathub org.kde.Platform/aarch64/5.15
error: Error searching remote flathub: Can't find ref org.kde.Platform/aarch64/5.15
```

Apparently flathub is not finding any repository:

(e.g. GIMP)
```
drace@nx:~$ flatpak install flathub org.gimp.GIMP
error: Error searching remote flathub: Can't find ref org.gimp.GIMP
```

mengchaoheng, 2022-10-23:
@gorghino 
```
1.  [sudo apt install flatpak ](https://flatpak.org/setup/Ubuntu)
2. flatpak install flathub org.kde.Platform/aarch64/5.15
3. flatpak remote-add --if-not-exists thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo
4. flatpak install thopiekar.eu org.mavlink.qgroundcontrol
```
after that, maybe restart your computer is need. Run qgc by:
```
flatpak run --device=all org.mavlink.qgroundcontrol

```

mengchaoheng, 2022-10-23:
to install qgc on arm64 ubuntu, step-by-step is https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/

OliverHeilmann, 2022-11-06:
> @gorghino Now this method doesn't work because the author seems to have removed this repository, this method is come from [qgroundcontrol-on-flatpak](https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/)

Any updated solution without the repo availability? Do you know of any accessible clones/ forks? Thanks!

mengchaoheng, 2022-11-06:
> Any updated solution without the repo availability? Do you know of any accessible clones/ forks? Thanks!

sorry I don't know @OliverHeilmann

mengchaoheng, 2022-11-17:
@thopiekar can you help me ? why:
```
parallels@ubuntu-linux-20-04-desktop:~$ flatpak install  flathub org.kde.Platform/aarch64/5.15-21.08
Looking for matches…
error: Nothing matches org.kde.Platform in remote flathub
parallels@ubuntu-linux-20-04-desktop:~$ flatpak remotes
Name         Options
flathub      system
thopiekar.eu system
parallels@ubuntu-linux-20-04-desktop:~$
```

mengchaoheng, 2022-11-17:
@OliverHeilmann @gorghino  and @ everyone who need to run qgc on arm-based ubuntu: [summary as](https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/):

1. Must be install flatpak 1.12.4 or later. [To install it](https://flatpak.org/setup/Ubuntu), run the following in a terminal:
```
       sudo add-apt-repository ppa:flatpak/stable
       sudo apt update
       sudo apt install flatpak
```
2. `flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo`
3. `flatpak install flathub org.kde.Platform/aarch64/5.15`
4. `flatpak remote-add --if-not-exists thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo`
5. `flatpak install thopiekar.eu org.mavlink.qgroundcontrol`
   after that, maybe restart your computer is need.

gorghino, 2022-11-17:
Great!! Thank you @mengchaoheng

NCJ-T, 2023-07-21:
@mengchaoheng on Jetson nano

AliceJin908, 2024-02-20:
Hi guys, I got the same issue with the appimage file downloaded from the doc guide. Why don't we have an update on the appimage file OR the packages required before running the appimage?

AliceJin908, 2024-02-21:
> @OliverHeilmann @gorghino and @ everyone who need to run qgc on arm-based ubuntu: [summary as](https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/):
> 
>     1. Must be install flatpak 1.12.4 . To install it 1.12.4, run the following in a terminal:
> 
> 
> ```
> sudo add-apt-repository ppa:alexlarsson/flatpak
> sudo apt update
> sudo apt install flatpak
> ```
> 
>     2. `flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo`
> 
>     3. `flatpak install flathub org.kde.Platform/aarch64/5.15-21.08`
> 
>     4. `flatpak remote-add --if-not-exists thopiekar.eu https://dl.thopiekar.eu/flatpak/_.flatpakrepo`
> 
>     5. `flatpak install thopiekar.eu org.mavlink.qgroundcontrol`
>        after that, maybe restart your computer is need.

Thanks for this. QGC runs well now on ubuntu 22.04.3 (LTS) running on a Rpi4B but without video. Do you have the same issue?

AliceJin908, 2024-02-27:
> @OliverHeilmann @gorghino and @ everyone who need to run qgc on arm-based ubuntu: [summary as](https://thopiekar.eu/uav/qgroundcontrol-on-flatpak/):
>
>     1. Must be install flatpak 1.12.4 . To [...]

@mengchaoheng Thanks for sharing. I got QGC running on arm device now but without video. It connects to pixhawk hardware and shows all the data as normal, just without any video display. Did you get the same problem? Thanks

mengchaoheng, 2024-02-27:
Sorry I don't have any suggestions

OliverHeilmann, 2024-02-27:
> @mengchaoheng Thanks for sharing. I got QGC running on arm device now but without video. It connects to pixhawk hardware and shows all the data as normal, just without any video display. Did you get [...]

Hi @AliceJin908, hope your project is going well! In response to the above, are you absolutely sure that the video is not working because of the QGC install process? Just looking at the [PX4 Documentation](https://docs.px4.io/main/en/companion_computer/video_streaming.html) I can see that you need to install GStreamer (though you might already have this).

Second thing is to check if you have [connected the video feed](https://docs.px4.io/main/en/sim_gazebo_classic/#how-to-view-gazebo-video) to QGroundControl properly as I've linked.

Let me know how these suggestions go and good luck!