mavlink/qgroundcontrol #10206: QGC appears to open/take control of USB Serial interfaces it isn't configured to use

spiderkeys, 2022-03-10:
## Expected Behavior
If there is not a CommLink automatically configured to use a serial device, nor a box selected under General settings to automatically connect via `Pixhawk`, then that serial device should never be opened by QGC.

## Current Behavior
- Opened QGC with two serial interfaces connected on host device:
  - /dev/ttyUSB0 (USB serial adapter being used to monitor PX4 debug output on console pins via `picocom`)  
  - /dev/ttyACM0 (USB connection to FMU intended for use by QGC
 - QGC opened /dev/ttyUSB0 despite all auto connections being unchecked and no comm links specified for it (there was one specified for manual, non-auto connection to /dev/ttyACM0
 - Garbage starts spewing out of the console, almost as if the flight controller has been reconfigured for a different baud rate by QGC
 - Upon trying to close and re-open picocom, I am informed `Device or resource busy` as /dev/ttyUSB0 is being held by QGC
 - Close QGC
 - Re-open picocom and the device is available and comms are back to normal

## Steps to Reproduce:
Steps are provided above, but the strangest thing is that the behavior appears to be random. Sometimes it happens, sometimes it doesn't.

Out of 10 tries, 7 didn't cause the issue and 3 did.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: 4.2.0
- QGC build: Daily AppImage from a few weeks/months ago
- Flight Controller: Matek H743 Wing v1
- Autopilot (with version): PX4 master

MaEtUgR, 2022-03-16:
You might have already done this but for reference. If you don't turn of these auto connection options QGC will capture serial connections.
[image]

spiderkeys, 2022-03-16:
Thanks, the autoconnection box is unchecked for the one serial comm link I have defined (ttyACM0). I have a separate serial adapter (ttyUSB0) which is connected to the console uart pins on my FMU, which is the interface seeing the issue. There is no comm link entry in QGC for ttyUSB0, in this case. 

Similarly, all of the auto connection boxes in the QGC general settings are also disabled (pixhawk, sic, px4flow, etc):
https://docs.qgroundcontrol.com/master/en/SettingsView/General.html

zdanek, 2022-03-17:
If you confirm this as a bug and you will describe steps to reproduce and then expected behavior I can fix this as I'm currently making some of my tweaks around serial comm and know this piece of QGC code.

spiderkeys, 2022-03-17:
I can confirm now that it happens whether there are manually created Comm Links or not.

Settings:
[image]

Comm Links:
[image]

Picocom + nsh:
- nsh works fine and is responsive before opening QGC
- Open QGC and a few seconds later garbage is emitted in picocom
- nsh becomes unresponsive
[image]
- Close picocom and try to reopen (while QGC is still open), ttyUSB0 resource is busy
[image]
- Close QGC and ttyUSB0 is accessible again and nsh is responsive again
[image]

spiderkeys, 2022-03-17:
The FMU physical setup, for context:
[image]

mrpollo, 2022-03-23:
I have no experience with devices, have you seen anything like this before @DonLakeFlyer?

spiderkeys, 2022-03-23:
Can confirm that autoconnect is disabled across multiple QGC restarts, w.r.t. this issue.

DonLakeFlyer, 2022-03-23:
Someone will need to debug through this to figure out what is going on. Maybe RTK code trying to connect?

hhenryy, 2022-03-24:
This is perhaps not related, but I have seen similar behavior when I connect FMUv5 using USB and NXP UCAN32K  via serial2USB. There were some weird behavior on the NSH console of the NXP board. I will need to test again to make double sure.

zdanek, 2022-03-24:
I'm doing some RTK related changes (actually making RTK run with Android device :D) I think I can dig into it.

HTRamsey, 2025-09-03:
Do you know exactly which setting fixes it? I assume either enable automatic logging or enable automatic log uploads?