mavlink/qgroundcontrol #10207: android daily build stopped updating

tridge, 2022-03-11:
the android daily stopped updating on 17th Jan 2021. Anyone know how to track this down?

[image]
https://play.google.com/store/apps/details?id=org.mavlink.qgroundcontrolbeta

Davidsastresas, 2022-03-21:
Just to link the PR to this issue, so both can be closed when merged into master: #10208

I tested the PR locally and it works as expected

holden-zenith, 2022-05-12:
This can be closed

booo, 2023-01-07:
How does the playstore get updated?

Do we need to adopt the changes here https://github.com/mavlink/qgroundcontrol/pull/10208/files to the new github workflows?

Davidsastresas, 2023-01-07:
That PR, as far as I remember, was addressing the issue of the artifacts not being updated to github in actions. The issue of updating the .apk to the appstore wasn't addressed on that PR.

Latest update to Playstore was on Jan 27 2021. This one issue is solved ( CI is working and android artifacts are being uploaded to github actions ), so I think we can close it and continue the play store discussion in #10529

closed 2023-01-07 by Davidsastresas