mavlink/qgroundcontrol #10215: Improve Start/stop recording button

OrbitalTam, 2022-03-16:
Hello

I'm using QGround control with a Bluerobotics underwater ROV. I've mis-recorded multiple dives because it's not obvious enough when the footage is being recorded or not.

Is it possible to make it more visible if the program is recording the video footage? Like a flashing "recording" text or something else ? At the moment, I believe the only signs that the footage is being recorded is that the little red circle turns to square or vice versa on the livestream screen. When switching pilots for the rov it's unclear whether it's recording and leads to non recorded footage.

Many thanks
Thomas
Orbital Marine Power

diveexplorer, 2023-06-15:
There would be a huge improvement if the record button would change color when you push it… or/and recording time would be visual underneath, like counting time. Then it would be easier to see when you are recording and when you are not.

diveexplorer, 2023-10-27:
This improvement or update should still be worked on and fixed.

closed 2026-10-04 by github-actions[bot]