mavlink/qgroundcontrol #10237: Compile QGC on arm64 ubuntu.

mengchaoheng, 2022-03-31:
The following errors appear when compiling qgc, I guess it has something to do with the qgc compilation settings
```Console
root@ubuntu-linux-20-04-desktop:~/qgroundcontrol/build# qmake ../
Project MESSAGE: Qt version 5.15.2
Project MESSAGE: Linux build
Project MESSAGE: Daily Build
Project MESSAGE: GIT_DESCRIBE v4.2.0-108-g0261e8b46
Project MESSAGE: QGroundControl APP_VERSION_STR VERSION v4.2.0-108-g0261e8b46 4.2.0.108
Project MESSAGE: Release flavor
Project MESSAGE: Skipping support for Pairing
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Including support for AirMap
Project MESSAGE: Taisync disabled
Project MESSAGE: Microhard disabled
collect2 version 9.3.0
/usr/bin/ld.gold -plugin /usr/lib/gcc/aarch64-linux-gnu/9/liblto_plugin.so -plugin-opt=/usr/lib/gcc/aarch64-linux-gnu/9/lto-wrapper -plugin-opt=-fresolution=/tmp/ccFjuB6h.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc --build-id --eh-frame-hdr --hash-style=gnu --as-needed -dynamic-linker /lib/ld-linux-aarch64.so.1 -X -EL -maarch64linux --fix-cortex-a53-843419 -pie -z now -z relro /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/Scrt1.o /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/crti.o /usr/lib/gcc/aarch64-linux-gnu/9/crtbeginS.o -L/usr/lib/gcc/aarch64-linux-gnu/9 -L/usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu -L/usr/lib/gcc/aarch64-linux-gnu/9/../../../../lib -L/lib/aarch64-linux-gnu -L/lib/../lib -L/usr/lib/aarch64-linux-gnu -L/usr/lib/../lib -L/usr/lib/gcc/aarch64-linux-gnu/9/../../.. --version -lstdc++ -lm -lgcc_s -lgcc -lc -lgcc_s -lgcc /usr/lib/gcc/aarch64-linux-gnu/9/crtendS.o /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/crtn.o
Project MESSAGE: Including support for video streaming
GNU gold (GNU Binutils for Ubuntu 2.34) 1.16
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
sh: 1: /opt/Qt-arm64-5.15.2/bin/qmlcachegen: Exec format error
sh: 1: /opt/Qt-arm64-5.15.2/bin/qmlcachegen: Exec format error
sh: 1: /opt/Qt-arm64-5.15.2/bin/qmlcachegen: Exec format error
sh: 1: /opt/Qt-arm64-5.15.2/bin/qmlcachegen: Exec format error
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
WARNING: Failure to find: /root/qgroundcontrol/build/qgroundcontrol_qmlcache.qrc
WARNING: Failure to find: /root/qgroundcontrol/build/src_FirmwarePlugin_APM_APMResources_qmlcache.qrc
WARNING: Failure to find: /root/qgroundcontrol/build/src_FirmwarePlugin_PX4_PX4Resources_qmlcache.qrc
WARNING: Failure to find: /root/qgroundcontrol/build/src_Airmap_airmap_qmlcache.qrc
/opt/Qt-arm64-5.15.2/bin/rcc: File does not exist '/root/qgroundcontrol/build/qgroundcontrol_qmlcache.qrc'
/opt/Qt-arm64-5.15.2/bin/rcc: File does not exist '/root/qgroundcontrol/build/src_FirmwarePlugin_APM_APMResources_qmlcache.qrc'
/opt/Qt-arm64-5.15.2/bin/rcc: File does not exist '/root/qgroundcontrol/build/src_FirmwarePlugin_PX4_PX4Resources_qmlcache.qrc'
/opt/Qt-arm64-5.15.2/bin/rcc: File does not exist '/root/qgroundcontrol/build/src_Airmap_airmap_qmlcache.qrc'
root@ubuntu-linux-20-04-desktop:~/qgroundcontrol/build# qmake -v
QMake version 3.1
Using Qt version 5.15.2 in /opt/Qt-arm64-5.15.2/lib
root@ubuntu-linux-20-04-desktop:~/qgroundcontrol/build# uname -a
Linux ubuntu-linux-20-04-desktop 5.13.0-35-generic #40~20.04.1-Ubuntu SMP Mon Mar 7 09:21:14 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
root@ubuntu-linux-20-04-desktop:~/qgroundcontrol/build# 

CONTAINER ID   IMAGE                      COMMAND                  CREATED         STATUS         PORTS     NAMES
a7e81c80a108   carlonluca/qt-dev:5.15.2   "/bin/bash -l -c bash"   9 minutes ago   Up 9 minutes             naughty_lovelace
``` 
the [container](https://hub.docker.com/layers/carlonluca/qt-dev/5.15.2/images/sha256-58a8932d9ddc2dec5df62d2be00635b143a8f9f2830f326e7c051a4f32b23b05?context=explore) is run on docker. Almost all compatible qt5 containers have been tested, the same error.

mengchaoheng, 2022-04-02:
in new container [carlonluca/qt-dev:5.15.2_dev](https://hub.docker.com/layers/carlonluca/qt-dev/5.15.2_dev/images/sha256-14a3aaee8cb883476475dbacd5e1b66f43c08469cbc2c90091356b85d810cc70?context=explore). unfortunately, it failed. The output is :

<details><summary>CLICK ME</summary>
<p>

```Console

~ docker pull carlonluca/qt-dev:5.15.2_dev
5.15.2_dev: Pulling from carlonluca/qt-dev
57d0418fe9dc: Already exists 
de2c46ba14b1: Pull complete 
a7aa7e5cf4f2: Pull complete 
0b0e38ca7135: Pull complete 
Digest: sha256:d1b9dbfcd417372d800d1846251d87940d17c06f0d9f039e06b292acc7fcc1b7
Status: Downloaded newer image for carlonluca/qt-dev:5.15.2_dev
docker.io/carlonluca/qt-dev:5.15.2_dev
➜  ~ docker run -it --network=host carlonluca/qt-dev@sha256:14a3aaee8cb883476475dbacd5e1b66f43c08469cbc2c90091356b85d810cc70 /bin/bash
Unable to find image 'carlonluca/qt-dev@sha256:14a3aaee8cb883476475dbacd5e1b66f43c08469cbc2c90091356b85d810cc70' locally
docker.io/carlonluca/qt-dev@sha256:14a3aaee8cb883476475dbacd5e1b66f43c08469cbc2c90091356b85d810cc70: Pulling from carlonluca/qt-dev
Digest: sha256:14a3aaee8cb883476475dbacd5e1b66f43c08469cbc2c90091356b85d810cc70
Status: Downloaded newer image for carlonluca/qt-dev@sha256:14a3aaee8cb883476475dbacd5e1b66f43c08469cbc2c90091356b85d810cc70
root@docker-desktop:~# unmade -a
bash: unmade: command not found
root@docker-desktop:~# uname -a
Linux docker-desktop 5.10.104-linuxkit #1 SMP PREEMPT Wed Mar 9 19:01:25 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
root@docker-desktop:~# qmake -v
QMake version 3.1
Using Qt version 5.15.2 in /opt/Qt-arm64-5.15.2/lib
root@docker-desktop:~# ps
  PID TTY          TIME CMD
    1 pts/0    00:00:00 bash
   20 pts/0    00:00:00 ps
root@docker-desktop:~# ls     
root@docker-desktop:~# git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git
Cloning into 'qgroundcontrol'...
remote: Enumerating objects: 147475, done.
remote: Counting objects: 100% (859/859), done.
remote: Compressing objects: 100% (385/385), done.
remote: Total 147475 (delta 569), reused 669 (delta 467), pack-reused 146616
Receiving objects: 100% (147475/147475), 423.52 MiB | 10.65 MiB/s, done.
Resolving deltas: 100% (112513/112513), done.
Submodule 'libs/OpenSSL/android_openssl' (https://github.com/Auterion/android_openssl) registered for path 'libs/OpenSSL/android_openssl'
Submodule 'libs/eigen' (https://gitlab.com/libeigen/eigen.git) registered for path 'libs/eigen'
Submodule 'libs/libevents/libevents' (https://github.com/mavlink/libevents.git) registered for path 'libs/libevents/libevents'
Submodule 'libs/mavlink/include/mavlink/v2.0' (https://github.com/mavlink/c_library_v2.git) registered for path 'libs/mavlink/include/mavlink/v2.0'
Submodule 'libs/qmdnsengine' (https://github.com/patrickelectric/qmdnsengine) registered for path 'libs/qmdnsengine'
Submodule 'libs/qmlglsink/gst-plugins-good' (https://github.com/mavlink/gst-plugins-good.git) registered for path 'libs/qmlglsink/gst-plugins-good'
Submodule 'libs/xz-embedded' (https://github.com/Auterion/xz-embedded.git) registered for path 'libs/xz-embedded'
Submodule 'src/GPS/Drivers' (https://github.com/PX4/GpsDrivers.git) registered for path 'src/GPS/Drivers'
Cloning into '/root/qgroundcontrol/libs/OpenSSL/android_openssl'...
remote: Enumerating objects: 255, done.        
remote: Total 255 (delta 0), reused 0 (delta 0), pack-reused 255        
Receiving objects: 100% (255/255), 13.24 MiB | 3.45 MiB/s, done.
Resolving deltas: 100% (80/80), done.
Cloning into '/root/qgroundcontrol/libs/libevents/libevents'...
remote: Enumerating objects: 216, done.        
remote: Counting objects: 100% (216/216), done.        
remote: Compressing objects: 100% (110/110), done.        
remote: Total 216 (delta 91), reused 205 (delta 82), pack-reused 0        
Receiving objects: 100% (216/216), 74.83 KiB | 252.00 KiB/s, done.
Resolving deltas: 100% (91/91), done.
Cloning into '/root/qgroundcontrol/libs/xz-embedded'...
remote: Enumerating objects: 704, done.        
remote: Counting objects: 100% (704/704), done.        
remote: Compressing objects: 100% (279/279), done.        
remote: Total 704 (delta 372), reused 704 (delta 372), pack-reused 0        
Receiving objects: 100% (704/704), 125.18 KiB | 409.00 KiB/s, done.
Resolving deltas: 100% (372/372), done.
Cloning into '/root/qgroundcontrol/libs/qmdnsengine'...
remote: Enumerating objects: 1524, done.        
[... 762 lines cut ...]
…

compiling moc_SettingsFact.cpp
compiling moc_MAVLinkInspectorController.cpp
compiling moc_AirspaceManager.cpp
compiling moc_SubtitleWriter.cpp
compiling moc_VideoManager.cpp
compiling moc_GstVideoReceiver.cpp
compiling moc_VideoReceiver.cpp
compiling moc_gstqsgtexture.cpp
compiling moc_qtwindow.cpp
compiling moc_qtitem.cpp
linking staging/QGroundControl
echo Post Link Common && mkdir -p staging/Qt/libs && mkdir -p staging/Qt/plugins && cp -f --dereference /opt/Qt-arm64-5.15.2/lib/libQt5Charts.so.5 staging/Qt/libs/ && cp -f --dereference /opt/Qt-arm64-5.15.2/lib/libQt5Core.so.5 staging/Qt/libs/ && cp -f --dereference /opt/Qt-arm64-5.15.2/lib/libQt5DBus.so.5 staging/Qt/libs/ && cp -f --dereference /opt/Qt-arm64-5.15.2/lib/libQt5Gui.so.5 staging/Qt/libs/ && cp -f --dereference /opt/Qt-arm64-5.15.2/lib/libQt5Location.so.5 staging/Qt/libs/ && cp -f --dereference /opt/Qt-arm64-5.15.2/lib/libQt5Multimedia.so.5 staging/Qt/libs/ && cp -f --dereference /opt/Qt-arm64-5.15.2/lib/libQt5MultimediaQuick.so.5 staging/Qt/libs/ && cp -f --dereference /opt/Qt-arm64-5.15.2 [... cut ...]  /root/qgroundcontrol/deploy/qgroundcontrol-start.sh staging && cp -f /root/qgroundcontrol/deploy/qgroundcontrol.desktop staging && cp -f /root/qgroundcontrol/resources/icons/qgroundcontrol.png staging
Post Link Common
cp: cannot stat '/opt/Qt-arm64-5.15.2/lib/libicui18n.so*': No such file or directory
make: *** [Makefile:2564: staging/QGroundControl] Error 1
root@docker-desktop:~/qgroundcontrol/build# 

````

</p>
</details>

more information in: https://github.com/carlonluca/docker-qt/issues/1

mengchaoheng, 2022-04-02:
@DonLakeFlyer

HTRamsey, 2024-05-31:
https://doc.qt.io/qt-6/supported-platforms.html

closed 2024-05-31 by HTRamsey