mavlink/qgroundcontrol #10241: Feature-Discussion] Add Developer Mode to QGC

junwoo091400, 2022-04-04:
## Problem
As discussed in the last [PX4 Dev Call](https://discuss.px4.io/t/px4-dev-call-march-30-2022/26701), there was an opinion raised by @dagar on having a separate User QGC & Developer QGC, since each party requires different functionality / priorities.

## Trigger
After the latest PR to add a 'Reboot Vehicle' button to the main screen #10239 , which was definitely necessary / helping developers, Silvan raised a concern that it's too extreme for the users. When I thought about it, it totally made sense since a typical end-user probably never has to reboot a vehicle.

## Proposal
**Add a 'developer mode' to the QGC, as a settings (just like Android does)** so that certain special buttons like these can be added as a support only for developers.

## Discussion
There probably will be more features that can be added with creating this Developer mode. Up for some discussions!

WTPENGUIN, 2022-04-08:
I think there is a need to minimization parameter loading in 'USER QGC'. Normal users do not change the default parameter values ​​provided by the firmware (I think that the UART port setting is the limit of parameters that general users change). It can increase the connection speed between the vehicle and the QGC and avoid wasting time due to parameter loading failure.

In High Latency mode, QGC can view vehicle information without parameter loading (but missions are also unavailable). By applying this, it seem to be possible to create a user mode that does not load parameters, but can operate the vehicle.

HTRamsey, 2024-05-31:
There is the Advanced Mode. Please feel free to reopen if that is not sufficient.

closed 2024-05-31 by HTRamsey