mavlink/qgroundcontrol #10251: Qgroundcontrol daily Build 4.2 for android is failed upload mission in terrain frame

RookieNumber, 2022-04-09:
## Expected Behavior
Upload mission successfully to vehicle in terrain frame altitude mode

## Current Behavior
instead, the progress bar is running so quickly and when I download the mission from vehicle, I cant find the mission Im trying to upload. So it must be failed, but somehow, uploading mission in relative altitude mode is fine. Btw, it only happen in Android build, for windows build the mission upload process for all altitude mode is fine.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior.
1. compiling
2. I install it and running it on android 10
3. everything is oke, creating mission and connecting to vehicle is working fine.
4. set altitude mode in relative and upload it to vehicle is working fine.
5. then I create the same mission and set the altitude mode to terrain mode
6. when I upload it, it failed

## System Information
When posting bug reports, include the following information
- Operating System: [Android 10, Android 11]
- QGC Version: [4.2]
- QGC build: [self-built from the source.]
- Flight Controller: [pixhawk 1]
- Autopilot (with version): [Ardupilot Copter 4.0.7]

## Log Files and Screenshots
here's the log file, thank you 
[android.txt](https://github.com/mavlink/qgroundcontrol/files/8456119/android.txt)

mohammada315, 2022-04-12:
Hi, I also had this problem

RookieNumber, 2022-04-19:
> Hi, did you figure out the solution yet? or maybe a hint for this problem

mohammada315, 2022-04-20:
Yes , i changed this code
My problem is temporarily solved

`switch (_cameraCalc.distanceMode()) {
    case QGroundControlQmlGlobal::AltitudeModeRelative:
        mavFrame = MAV_FRAME_GLOBAL_TERRAIN_ALT;
        break;
    case QGroundControlQmlGlobal::AltitudeModeAbsolute:
    case QGroundControlQmlGlobal::AltitudeModeCalcAboveTerrain:
        mavFrame = MAV_FRAME_GLOBAL;
        break;
    case QGroundControlQmlGlobal::AltitudeModeTerrainFrame:
        mavFrame = MAV_FRAME_GLOBAL_TERRAIN_ALT;
        break;
    case QGroundControlQmlGlobal::AltitudeModeMixed:
    case QGroundControlQmlGlobal::AltitudeModeNone:
        qCWarning(TransectStyleComplexItemLog) << "Internal Error: _buildAndAppendMissionItems incorrect _cameraCalc.distanceMode" << _cameraCalc.distanceMode();
        mavFrame = MAV_FRAME_GLOBAL_TERRAIN_ALT;
        break;
    }`

mohammada315, 2022-04-20:
TransectStyleComplexItem.cc Line 1287

RookieNumber, 2022-04-20:
what happen

RookieNumber, 2022-04-20:
what happen with line 1287, the code you write located in  line 1239

RookieNumber, 2022-04-20:
Oke, I think this is work fine, for now, thank you so much

closed 2022-04-20 by RookieNumber