mavlink/qgroundcontrol #10254: Anybody was succesful in compiling on Apple silicon M1?

highfreq, 2022-04-12:
Anybody was succesful in compiling on Apple silicon M1?

mengchaoheng, 2022-04-19:
another issues maybe help you. 

> https://github.com/mavlink/qgroundcontrol/issues/10205
and

> https://github.com/mavlink/qgroundcontrol/issues/10237
There are some problems with qgc running and compiling on arm ubuntu. The key to the compilation problem is that qt has poor support for arm64 architecture. So I think macOS is similar.

ijulanday, 2022-09-08:
I was able to compile and run today! The main thing (I think) was that I kept getting a deprecated warning for the use of the Eigen libary, so I added `CONFIG += c++14` and `CONFIG -= c++17` to the qgroundcontrol.pro file. 

I also had some issues with the libevents library, the way the nlohmann json library was included was kinda weird so I had to change that as well. 

I'm on a 2020 M1 Air with Monterey 12.4

mengchaoheng, 2022-09-08:
> I was able to compile and run today! The main thing (I think) was that I kept getting a deprecated warning for the use of the Eigen libary, so I added `CONFIG += c++14` and `CONFIG -= c++17` to the [...]

can you summary your step so the developers don't need to find the guidelines

ijulanday, 2022-09-08:
I tried making a [pull request](https://github.com/mavlink/qgroundcontrol/pull/10405)!

Looks like I didn't make the submodule changes correctly though, all the automated checks fail.

hendrik42, 2023-09-17:
Just compiled ToT with no problems. Ready to close?

DonLakeFlyer, 2024-02-21:
I've been compiling on M1 with the new master which is on Qt 6 got months now. Dev guide is up to date with steps you need for Qt 6.

closed 2024-02-21 by DonLakeFlyer