mavlink/qgroundcontrol #10259: Geofence / Rally

flybrianfly, 2022-04-21:
I have a [flight control system](https://github.com/bolderflight/spaaro) and [MAV Link library](https://github.com/bolderflight/mavlink), which wraps around the C MAV Link implementation. Reporting autopilot as MAV_AUTOPILOT_GENERIC with the following capabilities:
   * MAV_PROTOCOL_CAPABILITY_PARAM_FLOAT
   * MAV_PROTOCOL_CAPABILITY_MISSION_INT
   * MAV_PROTOCOL_CAPABILITY_MAVLINK2
   * MAV_PROTOCOL_CAPABILITY_MISSION_FENCE
   * MAV_PROTOCOL_CAPABILITY_MISSION_RALLY

Using QGC v4.1.5, I'm able to upload fence vertices and rally points to my autopilot. Upon upgrading to v4.2.0, I can only upload mission plans and the fence and rally tabs are shown, but greyed out and non-clickable. On Windows 10.

## Expected Behavior
Expect to be able to upload fence vertices and rally points.

## Current Behavior
Fence and rally tabs are shown, but greyed out and non-clickable.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: 4.2.0
- QGC build: stable
- Flight Controller: custom / generic

mnumanuyar, 2022-08-31:
"greyed out" generally caused by qmls enabled property which in this case controlled in by 
```js
                    QGCTabButton {
                        text:       qsTr("Fence")
                        enabled:    _geoFenceController.supported
                    }
```
and geoFenceController.supported is at [geofencecontroller.cc](https://github.com/mavlink/qgroundcontrol/blob/43d412cbf11edd7d6577a5bd14bbd3965c7f948c/src/MissionManager/GeoFenceController.cc)
```cpp
bool GeoFenceController::supported(void) const
{
    return (_managerVehicle->capabilityBits() & MAV_PROTOCOL_CAPABILITY_MISSION_FENCE) && (_managerVehicle->maxProtoVersion() >= 200);
}
```
however this part is not changed for 5 years. So this shouldnt be.

As a sarting point, we could look at the logs.

Vehicle geofence capabilities are logged [here](https://github.com/mavlink/qgroundcontrol/blob/43d412cbf11edd7d6577a5bd14bbd3965c7f948c/src/Vehicle/Vehicle.cc#L1285)

Can you try again, in both versions, with vehicleLog open, and upload the logs here ?

booo, 2022-11-06:
No feedback from the user and likely a implementation problem in the flight control system.

flybrianfly, 2022-11-15:
@mnumanuyar, I had a flight control system set up to test this and get the logging data today. The issue is still present. Attached are logs from v4.1.5 (log_v415) and v4.2.4 (log_v424). It looks like v4.1.5 successfully sees the autopilot capabilities messages that we are sending and v4.2.4 does not. Please let me know what additional information I can provide while I have this set up.
[log_v415.txt](https://github.com/mavlink/qgroundcontrol/files/10008031/log_v415.txt)
[log_v424.txt](https://github.com/mavlink/qgroundcontrol/files/10008032/log_v424.txt)

flybrianfly, 2022-11-15:
@booo, please re-open, this is still an issue. It took us a bit to set the test back up.

Yatin-Khurana, 2022-11-15:
You must be using mavlink 2 in latest qgc to enable fence and rally points, there's a peculiar condition, you have to first power the vehicle and then start the QGC, most prob a bug in qgc

flybrianfly, 2022-11-15:
The vehicle was started and running first in both of these tests. I'm reporting v2 support with the MAV_PROTOCOL_CAPABILITY_MAVLINK2 flag and in the v4.1.5 log file, you can see that QGC sees mavlink v2 as supported on line 22 as "[D] at C:\projects\qgroundcontrol\src\Vehicle\Vehicle.cc:1188 - ""Vehicle supports Mavlink 2.0"". It's weird, looking at the v4.2.4 log file, it's like QGC isn't picking up any of the capabilities flags that I'm sending.

Yatin-Khurana, 2022-11-15:
Is this a modified fork you are using ? Try on the latest stable branch

flybrianfly, 2022-11-15:
It's the latest stable on Windows 10. The older v4.1.5 is downloaded from the [GitHub release](https://github.com/mavlink/qgroundcontrol/releases/tag/v4.1.5).

booo, 2022-11-15:
Did you investigate the reason for the timeout messages? As far as I can see there are timeouts for the request of the AUTOPILOT_VERSION ([ #148 ](https://mavlink.io/en/messages/common.html#AUTOPILOT_VERSION)):

```
...
"_waitForMavlinkMessage msg:timeout 148 1000"
...
```

Maybe you should enable logging for `InitialConnectStateMachineLog` too to get a better overview about what is happening.

Some pointers to your code where you implement the initialization of your FC would be nice too.

flybrianfly, 2022-11-16:
Thanks for the hint. Based on the output from the log, it looks like older versions of QGC were using the [MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES) message and newer versions are only using the [MAV_CMD_REQUEST_MESSAGE](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_MESSAGE) to request the autopilot capabilities. Our flight software was responding to the now deprecated MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES message, but not the newer MAV_CMD_REQUEST_MESSAGE.

closed 2022-11-16 by flybrianfly

flybrianfly, 2022-11-16:
Just wanted to confirm that I added the appropriate response to the MAV_CMD_REQUEST_MESSAGE and fence / rally point options are now available in QGC v4.2.4. Thanks again.