mavlink/qgroundcontrol #10260: No instructions for building for Airmap

hamishwillee, 2022-04-21:
The [build instructions for Linux](https://dev.qgroundcontrol.com/master/en/getting_started/) fail because they assume Airmap SDK is present, but there are no instructions for getting the Airmap dependencies. 

How do we get/add the airmap dependencies? 

These should be added to https://dev.qgroundcontrol.com/master/en/getting_started/#install-qt step 4 "Install Optional/OS-Specific Functionality"

ckpradip, 2022-05-16:
@hamishwillee , do we have any updates regarding this ?
I created a file `user_config.pri` in project root directory and added this line - `DEFINES += DISABLE_AIRMAP`. But I am still seeing the same error. I am not sure if I am missing anything else !

Kim-zhi-jiang, 2022-06-13:
echo "CONFIG = WarningsAsErrors" >> /home/xxx/qgroundcontrol/user_config.pri
echo "DEFINES = DISABLE_AIRMAP" >> /home/xxx/qgroundcontrol/user_config.pri

closed 2024-12-01 by HTRamsey