mavlink/qgroundcontrol #10269: Unable to download (and flash) firmware

AndKe, 2022-04-27:
## Expected Behavior
download & flash

## Current Behavior
(tested on current master:)
```
 ./qgroundcontrol-start.sh 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Settings location "/home/andre/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/andre/Documents/QGroundControl Daily/Logs"
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
Map Cache in: "/home/andre/.cache/QGCMapCache300" / "qgcMapCache.db"
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
QGCPositionManager error 0
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
Download QGC stable version failed "Error during download. Error: 6"
"Hex ProfiCNC" "CubeOrangeRevB-BL"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
FirmwareUpgradeLog: PX4 releases github download failed "Error during download. Error: 6"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
FirmwareUpgradeLog: ArduPilot Manifest download failed "Error during download. Error: 6"

```

## Steps to Reproduce:
1. try to flash firmware
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: ubuntu 22.04
- QGC Version: master (27april))
- Flight Controller: Cube Orange
- Autopilot (with version): Plane 4.1.7(current stable)

## Detailed Description
n/a

## Log Files and Screenshots
see above

sq6emm, 2022-07-13:
Same problem here, same Linux, pixhwak 6x.

AndKe, 2022-08-19:
qrc:/qml/QGroundControl/ArduPilot/APMSensorIdDecoder.qml:121: TypeError: Cannot read property 'valueString' of null
qrc:/qml/QGroundControl/ArduPilot/APMSensorParams.qml:103: ReferenceError: insIdParamsAvailable is not defined
"Hex ProfiCNC" "CubeBlack"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
FirmwareUpgradeLog: PX4 releases github download failed "Error during download. Error: 6"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
FirmwareUpgradeLog: ArduPilot Manifest download failed "Error during download. Error: 6"
"3D Robotics" "PX4 BL FMU v2.x"
setCurrentPlanViewSeqNum

mkhuthir, 2022-09-18:
Same here:
OS: Ubuntu LTS 22.04
FC: Pixhawk 4

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Settings location "/home/mkhuthir/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/mkhuthir/Documents/QGroundControl/Logs"
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
Map Cache in: "/home/mkhuthir/.cache/QGCMapCache300" / "qgcMapCache.db"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
qml: QGCCorePlugin(0x55b73d6f55e0) []
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
Download QGC stable version failed "Error during download. Error: 6"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
"3D Robotics" "PX4 FMU v5.x"
FirmwareUpgradeLog: PX4 releases github download failed "Error during download. Error: 6"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
FirmwareUpgradeLog: ArduPilot Manifest download failed "Error during download. Error: 6"
"ArduPilot" "Pixhawk4-bdshot-BL"

AndKe, 2022-09-28:
Do you all have git set up with github certs up on the computers in question?
@DonLakeFlyer  - please give us a hint about what's going on here..

mkhuthir, 2022-09-28:
Yes I do have github and gh CLI with certs installed, but issue was the same. I was able to resolve it only by downgrading ubuntu from LTS 22.04 to LTS 20.04. It seems issue is related to QT support in LTS 22.04. When I downgraded to LTS 20.04 below errors stopped appearing:

qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate

AndKe, 2022-09-28:
Maybe a newer QT version fixes it? I'm on 5.15.2

mkhuthir, 2022-09-28:
Maybe, I cant test this now since my system already downgraded to LTS 20.04.

DonLakeFlyer, 2022-09-28:
No idea. If I had to guess something changed on the server side with respect to https certs.

AndKe, 2022-10-31:
BTW: Ubuntu 22.10 has the same issue. And I have no other app than QGC and AP2 that have such issues.

ardugitter, 2022-11-12:
I am seeing this with the Appimage as of today. Last time I updated firmware was a few weeks ago so the problem has appeared since then.

steamfoundry, 2022-12-09:
Same here on Ubuntu 22.10 with a telemetry radio

kibidev, 2023-03-08:
I see this for various HTTPS operations in Ubuntu 22.04 as well.
From what I see, qt 5.15.2 does not seem to fully support openssl 3 libraries (which in included in 22.04, while e.g. 20.04 uses openssl 1.1.x).

https://wiki.qt.io/Qt_5.15.2_Known_Issues

kibidev, 2023-03-10:
For Android version:
I tested the newest release on my phone (Samsung Galaxy S9+/SM-G965F, Android 10).
The 32 bit version does not seem to work with HTTPS at all (get "TLS initialization failed"), while the 64 bit version works without any errors.

Tested this by selecting a map provider which communicates with HTTPS.

AndKe, 2023-03-14:
@kibidev good catch .. this means this could be worked around if QGC contained it's own OpenSSL 1.1.x  - this would be solved?

kibidev, 2023-03-14:
@AndKe I guess this would be a short term solution. Openssl 1.1.1 is end-of-life in 6 months, so QGC should support Openssl 3.0 as soon as possible. But this I think requires either qt 6, or 5.15 commercial version.

Ryanf55, 2023-04-14:
Is there a plan for this, or a workaround? 
(Linux Ubuntu 22.04 user here)

rtalans, 2023-04-14:
I opened #10647, sorry, in my weeks of searching for a solution, this one never popped up.
We are 13 days shy of 1 year since this issue was tagged as "Priority: Critical". I am new to this universe. Is that kind of response time to a "Priority: Critical bug normal?

Ryanf55, 2023-04-14:
> I opened #10647, sorry, in my weeks of searching for a solution, this one never popped up. We are 13 days shy of 1 year since this issue was tagged as "Priority: Critical". I am new to this universe. Is that kind of response time to a "Priority: Critical bug normal?

To be fair, it was only tagged critical on `Mar 8`.

rtalans, 2023-04-15:
@Ryanf55 
I didn't realize that. Thank you.

Hopefully it will get more attention now.

rtalans, 2023-04-15:
@Ryanf55 
2 more questions:

1. Why does it work when flashing PX4 Pro (instead of ArduPilot), or is this not a comm problem between the PC and the flight controller?
[image]

2. Is there a way to elevate this issue or is it in some queue?

kibidev, 2023-04-15:
> Is there a plan for this, or a workaround? (Linux Ubuntu 22.04 user here)
> 
As for a workaround, have you tried to manually download the firmware and select it from disk when flashing?

It should be fairly easy to hack this in the code as well, but as that would mean disabling security (not check https server certificates) I do not like it. A bit better fix is to add an explicit option to disable certificate checks. But it is still a work around.

As for why this takes a long time to fix, I would assume that it is because this is dependent on the qt version used, and hence no small task to fix in a proper way (the proper fix is to upgrade to e.g. qt 6.5).

DonLakeFlyer, 2023-04-15:
> As for why this takes a long time to fix, I would assume that it is because this is dependent on the qt version used, and hence no small task to fix in a proper way (the proper fix is to upgrade to e.g. qt 6.5).

It more because there aren't many people dealing with QGC maintenance any more. So bugs are kind laying around. There is ongoing help to make that better so hopefully things will improve.

That said I'll try to look at this when I get back home in 3 weeks. If you see nothing by then ping my to remind. There should be some workaround that can be made.

dHutchings, 2023-10-23:
@DonLakeFlyer hello!  I just wanted to reach out and see if this issue is still open?

I have been trying to get QGC to download a custom mapbox map that I made - and I had no luck with it.  neither with typing in a custom mabox via the URL option.

The command prompt spews "fetch tile error: "SSL handshake failed" and "QSslSocket: cannot call unresolved function SSL_get_peer_certificate" so I think it's the same issue as discussed above.  is there an ETA on this?

Or, alternatively, should I be buidling QGC on my computer from source for this?

Ubuntu 22.04 on a Galaxy Chromebook 2

AndKe, 2023-10-23:
FWIW: I just tested on the current master (built myself)  and it works fine (Ubuntu23.10)

DonLakeFlyer, 2023-10-30:
I believe this was likely fixed by the ubuntu openssl fix that went in a while back.

closed 2023-10-30 by DonLakeFlyer