mavlink/qgroundcontrol #10271: No QGC location with external GPS

Vigasto, 2022-04-28:
## Expected Behavior
Using external GPS device should provide GCS position and is shown on map

## Current Behavior
GCS position is not available even though the device is being read

## Steps to Reproduce:
1. Set NMEA GPS device to the correct source
2. (optional) if GPS is USB based, reconnect device
3. Restart QGC

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10, Ubuntu 20.04
- QGC Version: 4.2.0, 4.1.x
- QGC build: stable, self-built (4.1.x)
- Flight Controller: N/A
- Autopilot (with version): N/A

## Detailed Description
The issue is currently only experienced with v4.2.0 with v4.1.x still providing GCS location. File changes suggested that this is due to incorrect handling of stricter requirements in #9483 which enforces QGeoPositionInfo::HorizontalAccuracy or HDOP availability. Reverting these commits solves the issue.

 Also note that this requirements are also set only in follow me mode which is aimed at mobile device which has internal gps; maybe they need different handlings? 

## Log Files and Screenshots
- [Console Log](https://github.com/mavlink/qgroundcontrol/files/8580401/gps.log)
- GPS device is checked to have GNGSA messages as shown.
[image]

booo, 2022-08-22:
I found this gem in the [Qt documentation](https://doc.qt.io/qt-6/qnmeapositioninfosource.html#details):

> QNmeaPositionInfoSource supports reporting the accuracy of the horizontal and vertical position. To enable position accuracy reporting an estimate of the User Equivalent Range Error associated with the NMEA source must be set with [setUserEquivalentRangeError](https://doc.qt.io/qt-6/qnmeapositioninfosource.html#setUserEquivalentRangeError)().

Additional documentation on the setUserEquivalentRangeError Qt function: https://doc.qt.io/qt-6/qnmeapositioninfosource.html#setUserEquivalentRangeError

I added this little change to the current master code and the position update starts working again:

```diff
diff --git a/src/PositionManager/PositionManager.cpp b/src/PositionManager/PositionManager.cpp
index 10d26c3bb..53392b615 100644
--- a/src/PositionManager/PositionManager.cpp
+++ b/src/PositionManager/PositionManager.cpp
@@ -101,6 +101,7 @@ void QGCPositionManager::setNmeaSourceDevice(QIODevice* device)
     }
     _nmeaSource = new QNmeaPositionInfoSource(QNmeaPositionInfoSource::RealTimeMode, this);
     _nmeaSource->setDevice(device);
+    _nmeaSource->setUserEquivalentRangeError(5.1);
     setPositionSource(QGCPositionManager::NmeaGPS);
 }

```

booo, 2022-08-23:
Fixed in master.

closed 2022-08-23 by booo

Vigasto, 2022-09-05:
I see, no wonder the accuracy estimate weren't there. Thanks for the fix!

kevinjhur, 2023-03-23:
Thank you for solution.
I can see QGCicon on the flight view, but the icon move quickly to random location like flight.
It's like gps drifting.