mavlink/qgroundcontrol #10274: There is no speech or the speech is hoarse

boris-gu, 2022-05-01:
## Expected Behavior
After the launch, I should hear a speech about changing the state of the UAV.

## Current Behavior
There is no sound. If I install speech-dispatcher and espeak-ng, the speech will appear, but it will wheeze.

## Steps to Reproduce:
1. Download the latest AppImage file (daily or stable).
2. Make the file executable and run. Connect UAV and make sure there is no sound.
3. Close QGC. Install speech-dispatcher and espeak-ng and run QGC again
4. Connect the UAV and make sure that the speech is hoarse
5. To check that speech-dispatcher is actually working correctly and the problem is in QGC, run `spd-say "Hello World"` in the terminal

## System Information
When posting bug reports, include the following information
- Operating System: Manjaro Linux
- QGC Version: 4.2.0 or 4.1.7
- QGC build: stable or daily
- Flight Controller: It doesn't matter
- Autopilot (with version): It doesn't matter

## Detailed Description
I found out about the need to install speech-dispatcher when I launched QGC from the terminal.  
Output:
```
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Settings location "/home/boris/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(Russian, Cyrillic, Russia) ; Name "ru_RU" ; Preffered (used in maps):  "ru-RU"
LocalizationLog: Qt lib localization for "ru_RU" is not present
Error loading text-to-speech plug-in "speechd"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/boris/Документы/QGroundControl/Logs"
VideoReceiverLog: Stop called on empty URI
Map Cache in: "/home/boris/.cache/QGCMapCache300" / "qgcMapCache.db"
qml: QGCCorePlugin(0x55bf17808140) []
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
```
After the necessary installations, the output became this:
```
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Settings location "/home/boris/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(Russian, Cyrillic, Russia) ; Name "ru_RU" ; Preffered (used in maps):  "ru-RU"
LocalizationLog: Qt lib localization for "ru_RU" is not present
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/boris/Документы/QGroundControl/Logs"
VideoReceiverLog: Stop called on empty URI
Map Cache in: "/home/boris/.cache/QGCMapCache300" / "qgcMapCache.db"
qml: QGCCorePlugin(0x5645a61d0b30) []
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
_recalcFlightPathSegments homePositionValid false
_recalcFlightPathSegments homePositionValid false
```

## Log Files and Screenshots
Test. I run QGC, and switch modes to hear raspy speech. Then, in the terminal, I play a test speech to make sure that the problem is not in speech-dispatcher.
[https://youtu.be/ExOprytgLZo](https://youtu.be/ExOprytgLZo)

booo, 2022-11-06:
On my ArchLinux machine it sounds the same. I think that is the normal speech voice. Are you sure this is a bug?

booo, 2023-01-07:
Missing feedback from the user. Closing this.

closed 2023-01-07 by booo

sibujacob, 2023-06-10:
Any luck on resolving the No sound issue in QGC v 4.2.6