mavlink/qgroundcontrol #10280: Run fails on Ubuntu 22.04 with QT Creator

wikstorm, 2022-05-09:
## Expected Behavior
Followed exact steps on https://dev.qgroundcontrol.com/master/en/getting_started/index.html
Build passes but pressing play button to start application fails.

## Current Behavior
Output in QT:
```
14:12:05: Starting /home/carl/build-qgroundcontrol-Desktop_Qt_5_15_2_GCC_64bit-Debug/staging/QGroundControl...
QML debugging is enabled. Only use this in a safe environment.
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Settings location "/home/carl/.config/QGroundControl.org/QGroundControl Daily.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United States) ; Name "en_US" ; Preffered (used in maps):  "en-US"
Fact name  "offlineEditingCruiseSpeed" not found in ":/json/Units.SettingsGroup.json"
14:12:08: /home/carl/build-qgroundcontrol-Desktop_Qt_5_15_2_GCC_64bit-Debug/staging/QGroundControl exited with code 255
```

## System Information
```
uname -a
Linux xxxx-All-Series 5.15.0-27-generic #28-Ubuntu SMP Thu Apr 14 04:55:28 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

booo, 2022-11-06:
Does this still happen with newer builds of QGC?

booo, 2023-01-07:
Closing this due to missing feedback from user.

closed 2023-01-07 by booo