mavlink/qgroundcontrol #10281: Can I use Android Studio to modify QGC?

Zxr1230, 2022-05-09:
I want to modify QGC to fit my needs. I only have experience with building apps using Android Studio. I wonder if I am able to import the QGC source code as an Android Studio project so I can customize and build the new app. Thank you!

RishabhBhat2021, 2022-05-12:
I don't think you can use Android Studio to build QGC.
QGC is built upon Qt C++ Framework (hence the 'Q' in QGC), and will require Qt Creater to build it
Whereas, Android Studio is totally different framework (in Java/Kotlin).
 
If you are coming from Android Studio to Qt, then C++ is what you'll require
And as far as Qt Creator is concerned, you would be okay following the Qt and QGC documentation for the build

booo, 2022-09-13:
Question properly answered therefore I close the ticket.

closed 2022-09-13 by booo

danieffe, 2023-11-15:
If I used the comunity docker container to build QGC for Ubuntu 22.04, how can I build it for android and so make the apk file?