mavlink/qgroundcontrol #10287: Is there a firmware for the board with ID 3?

UnRuly000, 2022-05-12:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

QGroundControl can upgrade the firmware on Pixhawk devices, SiK Radios and PX4 Flow Smart Cameras.
All QGroundControl connections to vehicles must be disconnected prior to firmware upgrade.
Please unplug your Pixhawk and/or Radio from USB.
Found device: PX4 FMU V2
Connected to bootloader:
Version: 5
Board ID: 3
Flash size: 1032192
Downloading firmware...
From: C:/Users/admin/Downloads/px4fmu-v2_default.px4
Download complete
Downloaded firmware board id does not match hardware board id: 9 != 3
Image load failed
Upgrade cancelled
------------------------------------------
If upgrade failed, make sure to connect directly to a powered USB port on your computer, not through a USB hub. Also make sure you are only powered via USB not battery.
Upgrade cancelled
------------------------------------------

Is there a firmware for the board with ID 3?

WTPENGUIN, 2022-05-24:
Hardware board IDs are not determined by QGC. I know it is determined when building the PX4 firmware.
[Reference](https://github.com/PX4/PX4-Autopilot/blob/master/boards/px4/fmu-v2/firmware.prototype)

you may to ask questions to the PX4 community.

booo, 2022-11-16:
Question answered by fellow user.

closed 2022-11-16 by booo