mavlink/qgroundcontrol #10298: crashed of caused by map

jingwenyi, 2022-05-20:
Hi, all, I recently start to test qgroundcontrol . I find a bug of crashed,  the bug is very frequently. 

repeat steps:
I start qgroundcontrol app, at the same time,  qgroundcontrrol   links  by tcp with the simulator of my plane. Then I enter offlinemap setting and zoom in and zoom out, then I click the another button in application settings  frequently and repeatedly.  crashed is visible. I found this crashed in QGeoMapReplyQGC.cpp

void 
QGeoTileMapReplyQGC::timeout()
{
        if(_reply){
                // here crashed,  I delete this row, the crashed is not visible.
                 // _reply->abort();
         }
         emit aborted();
}

jingwenyi, 2022-05-20:
I see the latest code , the same bug is still.

bcanozter, 2023-03-21:
I experience this using CustomURL ` https://t2.openseamap.org/tile/{z}/{x}/{y].png ` . It causes frequent network reply errors, and app crashes as a result.

closed 2024-09-21 by HTRamsey