mavlink/qgroundcontrol #10306: Installation instructions for Ubuntu need additional line

zlite, 2022-05-28:
If you're installing QGC with Ubuntu 20.x under WSL, you need one more line in the installation instructions:

sudo apt-get install -y libpulse-dev

Arccoma, 2022-07-20:
> If you're installing QGC with Ubuntu 20.x under WSL, you need one more line in the installation instructions:
> 
> sudo apt-get install -y libpulse-dev

[image]

in my case(https://github.com/mavlink/qgroundcontrol/issues/10350), the map is broken.

droneOTTO, 2023-01-30:
what about Ubuntu 22.04?

I cant launch QGC, i keep getting qt.network.ssl: QSslSocket: cannot call unresolved function error.

please explicit explanation, I am new to linux and installing things.

DieBorr, 2023-05-28:
Hi @droneOTTO , I also had that problem some time ago, the problem was resolved at least for me when I changed to ubuntu 20.xx , If u need some explanation after some research I found that OpenSSL was one of the main problems, Actually I'm not very familiarized with OpenSSL but there is a problem with that, hope this can help you if you haven't found the problem.

enriquesbr, 2024-01-29:
I have similar issue  like yours my current version is openssl/jammy-updates,now 3.0.2-0ubuntu1.13 amd64 [installed] by any change do you know which version is required by Qgroundcostrol ?

closed 2024-05-31 by HTRamsey