mavlink/qgroundcontrol #10310: Mission Planning /Mission Flying Improvements

buckker, 2022-05-30:
Hi

I’m testing QGC with more depth since a few weeks (with a arducopter drone). Overall it’s really a nice piece of software. Thanks to everyone who work on the project. However, I see some improvements regarding the mission planner feature. 

**Make the current name of the loaded mission / KML file visible in QGC:** If I load a KML file or a flight mission, I can’t see the name of the loaded KML or mission in QGC. This feature would be an improvement for knowing which mission is actually loaded. If I do some modifications, I can update and overwrite the correct (loaded) mission file. With the actual version of QGC, I have no information about the loaded mission and saving to a wrong name can easily occur.  Ideally the missions and KML’s name would be displayed in the file window. The flight mission name could be also displayed in the top bar of the flight tab

**Flight area offset:** Add a parameter similar to turnaround distance to do a (negative) offset for the flight mission from the border of the green area (kml or done by tracing). With this improvement we can still see the correct area of the flight mission and modifying the trigger distance to reach the correct photo count. Sometimes it is necessary to offset the border lines because a forest is touching the field respectively the flight mission. Mission Planner add a similar feature during the last releases. 

**Prevent crossing border lines in survey:** I noticed that the survey feature sometimes crossing the border lines of the flight area. This could lead to crashes and unwanted flight paths. Mission Planner doesn’t have this behavior.

**QGC Android storage location:** Let the user set the storage location, for example a folder on the SD card of the device. Is it technically possible to change this on android?

Mission/KML naming:
[image]

Flight Area Offset: 
[image]

crossing borders:
[image]

Many thanks in advance

Kind Regards

Michael

booo, 2022-11-16:
Thanks for your feature request. Do you have the time to split these features up into individual tickets? I think that would help a lot.

flavves, 2022-11-17:
Hello, ı use qt QGC on qt ı wonder how can ı found missions drawing lines qml files.
[image]

buckker, 2022-11-18:
@booo Yes, I will open individual tickets for my requests.

HTRamsey, 2025-05-14:
closing since you made individual issues

closed 2025-05-14 by HTRamsey