mavlink/qgroundcontrol #10312: QGC + gstreamer deploying error with macdeployqt

PremNikhil, 2022-06-01:
**When I try to make an installer with QGC with CONFIG+= installer in qmake build additional arguments,**

**the QGCPostLinkInstaller.pri works and everything runs file till the macdeployqt, and macdeployqt throws this error :** 
error: no file at "/library/lib/gstreamer.framework/versions/1.0/lib"
**but the default path  for any framework  in macos islibrary/frameworks/ and for gstreamer :**
"/library/frameworks/gstreamer.framework/versions/1.0/lib"

**i created a lib folder in library and copied the gstreamer.framework to the :** "/library/lib/gstreamer.framework/versions/1.0/lib" 
**path to check if it solves the error
but then it throwed another error** 

error: /applications/xcode.app/contents/developer/toolchains/xcodedefault.xctoolchain/usr/bin/otool-classic: can't map file: /library/lib/gstreamer.framework/versions/1.0/lib (invalid argument)\n

**Can't find any solution anywhere please help!**

patrickelectric, 2022-06-01:
Can you inform how did you install gstreamer and which macos version are you running ?

MilanBacaCz, 2023-07-16:
have anyone found a solution?

zxlben, 2023-09-10:
This is a bug in the Qt tool macdeployqt: when processing the executable file import table, the tool only supports entries in the form of @rpath and frameworks and dylibs located in the Qt installation directory. Because when building QGroundControl, GStreamer's import table entry is special and uses the form starting with /Library/Frameworks, thus triggering the bug of macdeployqt.

To fix this problem, you need to download the code of macdeployqt, modify it, and replace the macdeployqt file in the Qt tool directory.
macdeployqt code: https://code.qt.io/cgit/qt/qttools.git/
Modification points:
src/macdeployqt/shared/shared.cpp
FrameworkInfo parseOtoolLibraryLine(const QString &line, const QString &appBundlePath, const QSet<QString> &rpaths, bool useDebugLibs)
{
    FrameworkInfo info;
    QString trimmed = line.trimmed();

    if (trimmed.isEmpty())
        return info;

    // Don't deploy system libraries.
    if (trimmed.startsWith("/System/Library/") ||
        (trimmed.startsWith("/usr/lib/") && trimmed.contains("libQt") == false) // exception for libQtuitools and libQtlucene
        || trimmed.startsWith("@executable_path") || trimmed.startsWith("@loader_path"))
        return info;

    // Resolve rpath relative libraries.
    if (trimmed.startsWith("@rpath/")) {
        QString rpathRelativePath = trimmed.mid(QStringLiteral("@rpath/").length());
        bool foundInsideBundle = false;
        foreach (const QString &rpath, rpaths) {
            QString path = QDir::cleanPath(rpath + "/" + rpathRelativePath);
            // Skip paths already inside the bundle.
            if (!appBundlePath.isEmpty()) {
                if (QDir::isAbsolutePath(appBundlePath)) {
                    if (path.startsWith(QDir::cleanPath(appBundlePath) + "/")) {
                        foundInsideBundle = true;
                        continue;
                    }
                } else {
                    if (path.startsWith(QDir::cleanPath(QDir::currentPath() + "/" + appBundlePath) + "/")) {
                        foundInsideBundle = true;
                        continue;
                    }
                }
            }
            // Try again with substituted rpath.
            FrameworkInfo resolvedInfo = parseOtoolLibraryLine(path, appBundlePath, rpaths, useDebugLibs);
            if (!resolvedInfo.frameworkName.isEmpty() && QFile::exists(resolvedInfo.frameworkPath)) {
                resolvedInfo.rpathUsed = rpath;
                resolvedInfo.installName = trimmed;
                return resolvedInfo;
            }
        }
        if (!rpaths.isEmpty() && !foundInsideBundle) {
            LogError() << "Cannot resolve rpath" << trimmed;
            LogError() << " using" << rpaths;
        }
        return info;
    }

    // Resolve /Library/Framewoorks libraries.
    if (trimmed.startsWith("/Library/Frameworks")) {
        QStringList parts = trimmed.split("/");
        // process as dylib
        if(trimmed.endsWith(".dylib")) {
            int part = 0;
            while (part < parts.count()) {
                const QString currentPart = parts.at(part).simplified() ;
                ++part;

                if (currentPart == "")
                    continue;
                if(part < parts.count()) {
                    info.frameworkDirectory += "/" + currentPart;
                } else {
[... 146 lines cut ...]
            info.version = currentPart;
            info.binaryDirectory = "Versions/" + info.version;
            info.frameworkPath = info.frameworkDirectory + info.frameworkName;
            info.frameworkDestinationDirectory = bundleFrameworkDirectory + "/" + info.frameworkName;
            info.binaryDestinationDirectory = info.frameworkDestinationDirectory + "/" + info.binaryDirectory;
            state = FrameworkBinary;
        } else if (state == FrameworkBinary) {
            info.binaryName = currentPart.contains(suffix) ? currentPart : currentPart + suffix;
            info.binaryPath = "/" + info.binaryDirectory + "/" + info.binaryName;
            info.deployedInstallName = "@executable_path/../Frameworks/" + info.frameworkName + info.binaryPath;
            info.sourceFilePath = info.frameworkPath + info.binaryPath;
            state = End;
        } else if (state == End) {
            break;
        }
    }

    if (!info.sourceFilePath.isEmpty() && QFile::exists(info.sourceFilePath)) {
        info.installName = findDependencyInfo(info.sourceFilePath).installName;
        if (info.installName.startsWith("@rpath/"))
            info.deployedInstallName = info.installName;
    }

    return info;
}

ryanjAA, 2023-09-21:
I wrote a how to on this a few days ago and it seemingly didn't get posted so lets try again...

You need to add the above to shared.cpp file in /QT/version/Src/qttools/src/macdeployqt/shared/shared.cpp

Just run a compare to see where to insert it.

Then open the macdeployqt.pro file. There are two so open the one in its folder one directory back.

Build it. It will put it in  /QT/version/clang_64/qttools/bin/

Copy the new macdeployqt binary above to /QT/version/clang_64/bin/

Quit and reopen QT Creator. Now it will work for gstreamer.

@hamishwillee as just an FYI for temporary purposes if people start asking re this.

zdanek, 2023-12-19:
This bug hits us hard, when compiling with gstreamer. 
Any other solutions that can be applied, without recompilation of Qt?

ryanjAA, 2023-12-19:
Most reliable solution I’ve found that lets us avoid specific toolchain issues is a centralized GitHub repo. Then just build via GitHub actions and always the same, always works with same level of granularity.

zdanek, 2023-12-19:
According to https://doc.qt.io/qt-6/macos-deployment.html
`macdeployqt` can be omitted if package is prepared "manually", that is, step by step. This might be best way to workaround macdeployqt problem.
I have only Ubuntu and Windows. I cannot help here. Anybody?

HTRamsey, 2024-05-31:
I don't believe this is still an issue in the cmake build, let's reopen if it shows up again.

closed 2024-05-31 by HTRamsey