mavlink/qgroundcontrol #10319: RFD900x: Params won't download and QGC ignores stop bits setting

JustThinkInC, 2022-06-05:
Currently I am unable to properly connect to the drone using QGC with the RFD900x despite no issues doing so on Mission Planner. QGC also seems to ignore the value I set for the stop bits (1) and keeps reverting to the default (2).
----

## Expected Behavior
QGC should download params when connected to UAV using RFD900x telemetry and maintain number of stop bits as 1

## Current Behavior
QGC cannot download params (Vehicle 1 did not respond) despite being connected. Comms stop bits is set at 2.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Create new comms connection for RFD900x using COM 3, baud 57600, and 1 stop bit
2. QGC connects but cannot download params
3. Editing the comms connection will show the number of stop bits as 2 instead of 1

## System Information
- Operating System: Windows 1
- QGC Version: 4.2.2
- QGC build: stable
- Flight Controller: Pixhawk 2.4.8
- Autopilot (with version): Ardupilot Copter 3.6.1

## Detailed Description
QGC cannot download params when using the RFD900x. It also does not respect the setting of the number of stop bits for the connection to the RFD radio. Despite me setting the stop bits to 1, when I edit the connection it shows as 2, I suspect this might be causing the problem (or part of it). I am using the point to point setup. I have tried changing the air speed, max window, toggling encryption on/off but no luck. Mission planner has no issue whatsoever regardless of my settings. 

## Log Files and Screenshots
[image]

fredowski, 2022-09-23:
The settings in RFD900 tools only configure the RFD900 modems but not the UART settings on the PC and the vehicle. You have in total 4 UARTS. The UART on your PC inside the FTDI cable, the UART at the RFD900 modem connected to your PC, the UART in the RFD900 modem at the vehicle and the UART on the Flight Controller. The UART settings in the flight controller are controlled via the "SERIAL_xxx" parameters.

JustThinkInC, 2024-11-25:
Coming back to this as I've had some more time to pick up the RFD900 again. I still cannot get it to work on QGC. Using Telem1 for the RFD900, Holybro Pixhawk 5X with ArduPilot 4. There is **0% packet loss** on both Mission Planner and QGC using the RFD900.
`SERIAL1_Baudrate` is set to` 57600`, `serial1 options` are `0`

I have also tried updating the radios to the [latest V4 beta:](https://rfdx.atlassian.net/servicedesk/customer/portal/1/article/452231169) but no change at all. I'm also now running the latest QGC release, no luck. I have tried building QGC on Windows 10, same issue no luck.

[image]

@DonLakeFlyer I see you have used the RFD in [the past ](https://github.com/mavlink/qgroundcontrol/issues/6330). Did you ever run into this issue?

DonLakeFlyer, 2024-12-02:
Nope, I haven't used a 900 in ages. When you say parameters won't download. In what way. For for info turn on ParameterManagerLog and post log here.

JustThinkInC, 2024-12-07:
Unfortunately this all that comes up: [qgclog.txt](https://github.com/user-attachments/files/18047842/qgclog.txt). I have tried disconnect, restart qgc, connect to see if more info would be logged but nope. RFDesign support has been unable to replicate the issue with my settings so I'm now tempted to believe I have some sort of weird hardware issue.

JustThinkInC, 2025-02-10:
Update: [RFD900X2-MultComm_V4.00B1.gbl](https://files.rfdesign.com.au/Files/firmware/RFD900X2-MultComm_V4.00B1.gbl) works and I no longer get this issue. Can confirm works well in flight too 😄

closed 2025-02-10 by JustThinkInC