mavlink/qgroundcontrol #10320: Deploying error :

PremNikhil, 2022-06-06:
**ERROR: no file at "/Library/lib/GStreamer.framework/Versions/1.0/lib"**

Getting this error while building qgc for mac with Installer config (building an installer file).

and when i open the .app file from the .dmg after the process is complete, its not working and showing that it can't some libraries like gst-net.1.0.0.dylib etc.

Please help! 

Couldnt find anything anywhere that helps!
thank you,

PremNikhil, 2022-06-06:
The path is supposed to be "Library/Frameworks/GStreamer.framework/Versions/1.0/lib"

Mac os : Monterey Version : 12.3.1

booo, 2022-11-16:
Can you check if b2887b43a187e310963ecb1276333e4f2a2877b4 fixed this problem?

closed 2024-05-31 by HTRamsey