mavlink/qgroundcontrol #10321: in custom build the name daily is appearing

RamananR-hash, 2022-06-12:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

ES-Alexander, 2022-06-15:
@RamananR-hash, you don't seem to have provided any information for your issue.

Going off your title alone, yes, custom builds tend to have the Daily name because they're not stable releases with a set numeric version number. I'm not familiar enough with QGC's codebase and build system to know how to change that, but it's also not unexpected behaviour, and is probably not worth attempting to bypass since it's a correct indicator that the build is a modified and not official version of QGC.

RamananR-hash, 2022-06-15:
Thank you for You're reply Mr.ES-alexander . I found some technique to remove it .fortunately it works .and this is my first time I am asking question on GitHub .and this is the first time i am getting reply .thank you for your time @ES-Alexander 

ANSWER IN QGC folder QGCCommon.pri file we have to delete the else statement from line 155 

and it works without any errors thank you

closed 2022-06-15 by RamananR-hash