mavlink/qgroundcontrol #10323: MAVLink folder dose not exist at ‘libs/mavlink/include/mavlink/v2.0’! Run ‘git submodule init && git submodule update’ on the command line.

Arccoma, 2022-06-17:
Hello, everyone.
I'm a newbie developer, unfamiliar with all three of Qt and QGC and Git.

[My first challenge]
https://dev.qgroundcontrol.com/master/en/getting_started/
A few days ago, I downloaded the qgc code with git clone according to the guide on the link above
Installation of Qt 5.15.2 and Visual Studio 2019 community (64 bit)
The build was successful.

Of course, I couldn't succeed at once, but after a few days of challenge, I did the following three things
I managed to succeed. (I was a little surprised because it took about 10 minutes to build. )
1. The system environment variable has jom added (C:/Qt/tools/QtCreator/bin/jom)
2. Unzip the source code to the C:/User/.... subfolders
3. The Appsetting.cpp file was changed to UTF-8 BOM and saved (using notepad++)
I only followed the method that was informed with the help of other developers, and I haven't understood the meaning of those three tasks yet, and I don't understand why that content is missing from the QGC guide.
It is not complete, but based on the above success experience, I proceeded with the next challenge.

 [ My second challenge ]
http://dev.qgc.dimianzhan.com/en/getting_started/
This time, I referred to the guide of the link above
Without downloading the source code through git clone,
https://github.com/mavlink/qgroundcontrol/releases/tag/v3.5.6 here
Download and extract the qgc v3.5.6.zip file
Install Qt 5.11.0 and Visual Studio 2015 community (32 bit)
I tried building. With the expectation that it'll work out at once.

However, contrary to expectations, less than a second later, I encountered the first error.
The error related to the submodule and the details are as follows.

[image]
Figure 1: Build Error Message

Project ERROR: MAVLink folder dose not exist at ‘libs/mavlink/include/mavlink/v2.0’!
Run ‘git submodule init && git submodule update’ on the command line.

I checked the error message and just downloaded and unzipped it
I checked the libs/mavlink/include/mavlink/v2.0 folder. And I was a little surprised to see that there was nothing in it and it was empty. 
"Why isn't there anything in the folder? Did someone accidentally miss it when I uploaded it to GitHub?"

After searching on the web for some time,
I found out that Git managed the project using 'submodule' for efficiency, and that when I downloaded the compressed file without using the git clone command, I deliberately left out the contents of 'submodule' and compressed it, so I got an empty folder. And [My First Challenge] confirmed that the 'submodule' folder of the same name downloaded with the git clone command is not empty and contains some files needed for the build.
And the information about the submodule that I need to fill is in the .gitmodules file
It was confirmed that the number of submodules was 2.

[image]
Figure 2: Submodule Information

Let's move on to the next sentence of the error message
“Run ‘git submodule init && git submodule update’ on the command line.”
In order to solve the problem, search is done without a sufficient understanding of git
I updated the submodule as below and filled in two empty folders.

[image]
Figure 3: submodule “src/GPS/Drivers”  update

[image]
Figure 4: submodule “libs/mavlink/include/mavlink/v2.0”  update

In each empty folder, four git commands were entered by opening a cmd window with Git Bash.
1. git init
2. git submodule init
3. git submodule add <submodule adress>
4. git submodule update --recursive

I saw an empty folder filled immediately after the add command.
But through the above work, MAVLink folder
It does not appear to occur in 'libs/mavlink/include/mavlink/v2.0'

I tried building again, hoping that the cause of the error was resolved through the above work.

[image]
Figure 5: Check the QT and compiler version

The result is
[image]
Figure 6:  Again, the same error message as the first time

I saw the same error message again.
It is judged that I did not perform the submodule update properly.
I don't know which part I've gone wrong yet.
I ask for your help.

## System Information
- Operating System: Windows10
- QGC Version: 3.5.6
- QGC build: zip file from https://github.com/mavlink/qgroundcontrol/releases/tag/v3.5.6
- Flight Controller: N/A
- Autopilot (with version): N/A

WTPENGUIN, 2022-06-17:
Is there any reason to use the old QGC 3.5.6 version?

Arccoma, 2022-06-18:
I don't know the exact version, but building a modified code based on the 3.x.x version is my current mission and final goal for my company.
The person who modified the code is someone I don't know and it is difficult to contact now and in the future.
The submodule is empty in the data he left behind, and of course, the build failed.
The information given to me is, at some point in the past,
It's just that he developed it based on the 3.x.x version, following the guidance of dev.qgc.dimianzhan.com/en/getting_started/.
I checked on the release page that there are a total of 30 versions of 3.x.x, and I chose the most recent version, 3.5.6
[My second challenge] is in progress.

Assuming that if the major versions are the same, the build environment is the same, I hope that if this challenge succeeds, I can achieve my final goal.

I liked Sherlock Holmes' mystery novels, but when I was about to become a detective, I felt that it was a tough job.

WTPENGUIN, 2022-06-18:
It looks like you need more knowledge of Git.

I just cloned the source code of QGC v3.5.6 and found nothing strange.

Try the following methods.

`git clone -b "Stable_V3.5.6" --single-branch https://github.com/mavlink/qgroundcontrol.git`
`cd qgroundcontrol/`
`git submodule update --init --recursive`

Kim-zhi-jiang, 2022-06-18:
make submodulesclean
git submodule update --init --recursive
make submodulesclean
git submodule update --init --recursive

Make sure to download them all, because you never check them

Arccoma, 2022-06-18:
I followed the magic spell you gave me and ordered the computer to get the source code including the submodule. And the build was successful. Thank you. ^^
Contrary to expectations that the wizard will complete the build in less than 10 minutes, I encountered an error related to 'rc' in a minute.

[image]
After hours of searching for a solution, I found out that it was a kind of compiler, and I was able to add it to the system environment variable Path to complete the build.

[image]
I found a total of 6 rc.exe under the C:/Program Fles(x86)/Windows Kits/10/bin folder, but I didn't know which rc to add, so I just chose the one with the highest number at the bottom.
I guess I was lucky, and then the build succeeded at once. But I'm not sure if it was the right choice yet.
[image]
I don't fully understand the magic spell yet, but I think it's related to the Git branch window on the left side of the Qt Creator window. I feel my lack of understanding of Git.
And this question arose.
Do I have to use Git instead of downloading the compressed file right away? Then, I wondered why compressed files can be received separately. I think it's a question caused by my lack of understanding of Git.

[Challenge for my final goal]
Taking advantage of my successful experience with the help of Penguin, I immediately challenged the final goal of building the qgc 3.x version. But I couldn't think of a way to apply Penguin's magic spell to this challenge.
So, I copied and pasted the contents of the submodule folder in v3.5.6 as it is in the empty submodule folder.
And in about a minute, the build failed.
It was not an error related to the submodule, and the problem that I thought was solved by adding 'rc' appeared again.
And this time, I met a new problem related to 'HEAD'.

> fatal: Not a valid object name HEAD
> fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
> Use '--' to separate paths from revisions, like this:
> 'git <command> [<revision>...] -- [<file>...]'
> fatal: Needed a single revision

I don't know if it's the same as that 'HEAD' displayed in the branch window.
[image]

I look forward to the help of Penguin and other seniors.

Arccoma, 2022-06-19:
> make submodulesclean git submodule update --init --recursive make submodulesclean git submodule update --init --recursive
> 
> Make sure to download them all, because you never check them

Thank you for your reply. Unfortunately, I don't think I can understand your connotation yet. I'll look for more for understanding.

WTPENGUIN, 2022-06-20:
@Arccoma 

If you need to build QGC 3.X version, update the submodule using the git command. Copying and pasting can cause unintended behavior.

복붙말고, 명령어를 통해 서브모듈 업데이트 한번 해보세요.

P.S 첨부하신 사진을 보니 한국인이신 것 같은데, 선임 개발자 분들께 깃이나 빌드 관련해서 도움 요청하는 것이 더 좋아 보여요.
(I'm not familiar with English, so I used Korean for quick help. Please excuse me.)

Arccoma, 2022-06-20:
> @Arccoma
>
> If you need to build QGC 3.X version, update the submodule using the git command. Copying and pasting can cause unintended behavior.
>
> 복붙말고, 명령어를 통해 서브모듈 업데이트 한번 해보세요.
>
> P.S 첨부하신 사진을 보니 [...]

만나서 반갑습니다.^^  저도 펭귄님 프로필 보고 그냥 한국말로 해도 되나 한참 고민했습니다.  짧은 영어 실력이라 번역기 돌리고 제대로 된건지 한 번 더 확인하고 하느라  빠른 소통을 할 수 없어 답답했습니다.  깃허브 정책이나 분위기를 아직 잘 몰라서 망설여지는데  같은 언어 사용자와는 꼭 영어를 쓰지 않아도 되는 것인가요?  
 
Good to meet you.^^ When I saw Penguin's profile, I was wondering if I could speak in Korean. I was frustrated that I couldn't communicate quickly because I was checking whether it was properly done after using the translator because of my short English skills. I'm hesitant because I don't know the GitHub policy or atmosphere yet, but do I not have to speak English with people who speak the same language other than English?

WTPENGUIN, 2022-06-20:
@Arccoma 

원래는 영어를 쓰는 것이 권장되지만, 저도 영어에 익숙하지 않다 보니 빠른 도움을 주기 위해 한국말 썼다. 이해 부탁 드린다. 식으로 했습니다.

영 마음에 걸리신다 하면, 제 프로필에 있는 메일로 따로 연락을 주는 방식도 괜찮다 봅니다.

Arccoma, 2022-06-20:
With the help of [WTPENGUIN](https://github.com/WTPENGUIN) and other people around me,
I completed [Challenge for my final goal ] in the following way. 

step1. The git clone -b command replicates a branch named "Stable_V3.5.6" (which is not yet fully understood but inevitably used) to the local PC.
step2. The empty submodule folder of the cloned branch was left unchanged for a while.
step3. All the contents of the project, believed to have been modified from the 3.x version left by the previous developer, 
          were copied and overwritten with v3.5.6 created in Step 1.
step4. The submodule update, which had been postponed for a while in the step2, was carried out.

The above method allows me to complete the build of the source code left by my predecessor and view the execution screen
There are still a number of questions as below.

The above method allows me to complete the build of the source code left by my predecessor and view the execution screen
There are still a number of questions as below.

> Is this correct or there are no errors in the build process (for example, whether 'rc.exe' was selected correctly)
> 
> There are 30 versions of 3.x on the release page, but why are there only 7 versions of 3.x in the branch?
> 
> Are all submodules of the same 3.x version the same content?
> 
> Is the 3.x version of the build environment correct for Qt 5.11.0 and Visual Studio 2015 32bit?
> 
> Is there any way to find out the exact version of the QGC left by my predecessor?
> 
> Is it normal to develop based on the old version that is two years old instead of the latest version?

I look forward to resolving the above questions as my understanding deepens.
If I find the answer, I will update the contents.
I am grateful to everyone.

danilopmaciel, 2022-07-18:
the sambe problem.
:-1: error: Project ERROR: MAVLink folder does not exist at 'libs/mavlink/include/mavlink/v2.0'! Run 'git submodule init && git submodule update' on the command line.

Arccoma, 2022-08-01:
> 1. The system environment variable has jom added (C:/Qt/tools/QtCreator/bin/jom)
> 2. Unzip the source code to the C:/Users/.... subfolders
> 3. The AppSetting.cc file was changed from UTF-8 to UTF-8 BOM encoding and saved (using notepad++)
> 
> I only followed the method that was informed with the help of other developers, and I haven't understood the meaning of those three tasks yet, and I don't understand why that content is missing from the QGC guide.

Your question suddenly came to me today.
What is the intention of this question?
I feel like I'm asking like this.
"Why are you doing such an unconventional thing? People usually use the latest version..."

Arccoma, 2022-08-01:
> Is there any reason to use the old QGC 3.5.6 version?

Your question suddenly came to me today.
What is the intention of this question?
I feel like I'm asking like this.
"Why are you doing such an unconventional thing? People usually use the latest version..."

yutian-9264, 2023-02-16:
`git clone --recursive https://github.com/mavlink/qgroundcontrol.git `
from
[https://diydrones.com/forum/topics/why-the-qgroundcontrol-is-so-error-when-i-build-this-source-code](url)

Davidsastresas, 2023-02-19:
I think we can close this one due to inactivity...

closed 2023-02-19 by Davidsastresas

sevahul, 2023-10-02:
I had the same issue.
The solution was to install cmake:
```
sudo apt install cmake
```
Then run `qmake ..` again, it should work